samtap/fang-hacks #282: change network settings

kamilio69, 2017-12-14:
Hi, I have just changed my modem and so the network settings for the cam connection are different : ssid and key security ; I just wonder how to change the network settings, using cloud mode , I don't know ?

ril3y, 2017-12-14:
@kamilio69 did you disable the cloud app?  If not you can start the camera press the button, wait for the voice then use my QR code generator software to reconfigure your network.

Try it here:
https://codepen.io/ril3y/full/gXyzmO/  

There is also a video on how it works.

Riley

kamilio69, 2017-12-15:
Hi Riley, thanks , it works, I see the right network settings on the fang hack script. but now I  have an issue with my wifi connection because the cam 's led first flashes yellow/blue, and after flashes blue and after stop flashing. if you have an idea?

kamilio69, 2017-12-15:
it s ok now, I can connect the cam on remote access, but curiously the led doesn't work,

DerMilchShake, 2018-02-09:
@ril3y I also want to change network settings. Should I press first the setup button, while the Sd card with the fang hack is in the camera, and scan then the qr code with wifi settings?
Or should I first remove the Sd card, then press the setup button, put the Sd card with fang hacks in camera and scan the qr code?

madmax2, 2021-04-30:
@DerMilchShake I think you remove the SD card, so you can you the cloud (mi home) app to change the network settings

hackgrid, 2022-12-12:
> @kamilio69 did you disable the cloud app? If not you can start the camera press the button, wait for the voice then use my QR code generator software to reconfigure your network.
> 
> Try it here: https://codepen.io/ril3y/full/gXyzmO/
> 
> There is also a video on how it works.
> 
> Riley

This tool is unfortunately broken with the last update!

On a comment of the Youtube video, "hal novemila" discovered: 

"
Hi Riley, thanks so much for your useful tool :) Unfortunately it didn't work with my Fang camera and I was wondering why as I was using an old firmware. By chance I had register to codepen the day after last Christmas so I was able to figure out how to look at the console and the code. Using my very limited programming knowledge I figured out a couple of things to fix with your code which I did in a fork. First thing is that the camera model selection fails as the model var is set after the conditional branch and not before. As result the generated QR code is always for the Wize model and never for the Fang. Another little glitch is that the password is not fully "set" until the user click outside the password text input box. That can be confusing because people may try multiple passwords without clicking outside the input box therefore their QR code will never have the correct password but missing the last char. Regards.
"

So if you are wondering, why your camera does not connect to WIFI anymore, using this tool, thats why!

MrDeon, 2022-12-17:
> I made a fixed version of this tool here:
> 
> https://codepen.io/hackgrid/full/dyKEdeX

Thank you so much!! My Fang camera somehow forgot my SSID and no matter how many times I updated wpa_supplicant.conf, it wouldn't stick. I stumbled across the QR code page, but that didn't work either - however your update most certainly did. This fixed it perfectly!