samtap/fang-hacks #283: I found a way to downgrade the 3.0.3.56 firmware

Pancts, 2017-12-15:
Sorry I can only speak Chinese.

我的小方相机的固件是 3.2.0.30 , 无法使用hack, 我尝试了很多方法进行降级都失败了.
今天我在使用米家app时提示我升级固件到 3.3.3.25, 在升级的过程中, 我对小方相机进行了断网的操作.
之后米家app提示我升级失败, 是否重新升级.
在点击升级时, 我发现升级版本变成了 升级到 "3.0.3.56" 升级成功后我确认固件版本变成了3.0.3.56, 并且成功安装了hack.

很感谢samtap开发的hack程序, 他让我的小方可玩性更高了, 所以我把我的发现提交到这里, 不一定保证成功.

google translate:
My fang camera has a firmware of 3.2.0.30, can not use hack, and I tried many ways to downgrade.
Today, when I was using My Home app, I was prompted to upgrade the firmware to 3.3.3.25. During the upgrade process, I disconnected the camera's network.
After the rice home app prompts me to upgrade failed, whether to re-upgrade.
When I clicked Upgrade, I found that the upgrade version has been upgraded to "3.0.3.56." After the upgrade was successful, I confirmed that the firmware version has changed to 3.0.3.56, and hack was successfully installed.

Note: This operation I can not reproduce, does not guarantee success

pluskal, 2017-12-22:
Hi!, 

I have tried that with one cam that was on version 3.2.0.30. Sadly, it went through the update process... and now it is on the version 3.3.3.25.

My approach - start the update, wait until app showed 5% and disconnect network uplink from my wifi AP. Update still showed 5% so I have plugged network back an waited until 10% and disconnect it again. After about 10minutes, the app showed update fail so I have connected the network and suddenly, the camera showed that it is on version 3.3.3.25.

So be careful with this approach.

udivankin, 2018-02-11:
Followed these steps and got success.
https://www.youtube.com/watch?v=mpzPWYONWZA
Firmware https://www.dropbox.com/s/kzyxjdpxmyk...

1. Download and copy FIRMWARE_660R.bin to formatted SD, insert into the camera
2. Disconnect the camera from power source
3. While holding SETUP button, connect the camera to power source
4. Release the SETUP button after 10 seconds.
5. LED will first be steadily yellow, then blink yellow-blue after which you can reboot the camera (reconnect the power source)

pluskal, 2018-02-12:
@udivankin which firmware version you have previously installed?

udivankin, 2018-02-12:
@pluskal I believe 3.2.0.30