samtap/fang-hacks #284: Need help understanding Serial Port connection

Permensugus, 2017-12-20:
So, I think I broke something within the cam. It stopped working and only show orange LED. I can access the cam via serial port. I was thinking of reflashing the firmware, but when I login, I only see 1 folder, etc_default. Can anybody point a way to access sd card? I have been looking around the filesystem, but can't seems to find any media folder suggested. Thanks beforehand and sorry for noobish question!

samtap, 2017-12-20:
After logging in you're in /root which contains a folder named etc_default. The sdcard partitions are at /media/mmcblk0p1 and /media/mmcblk0p2

Permensugus, 2017-12-20:
thanks!! can I ask something else? What command can we use to edit file using terminal? Vi,nano,vim isn't installed, and I can't think of a way to do so. thanks!

samtap, 2017-12-21:
vi is available on the sdcard (run 'source /media/mmcblk0p2/data/etc/profile' if you login via telnet, not needed if you use ssh)

mccanos, 2018-01-19:
Hello, @Permensugus. I have the same problem and I want to know if you have been able to solve it and if you can help me with the steps needed to follow to solve this issue. I would like to know how can i go inside /media/mmcblk0p1 and /media/mmcblk0p2 as I only can see /root directory. by the way I also would like to know how have you been able to edit the files as I use putty on windows and all the text editors are available for Linux.... Thanks in advance and best regards.

Permensugus, 2018-01-20:
@mccanos , it quite silly actually, when you first logged in, you are inside the root folder, so you just have to do "cd .." or "cd /" and you can see the whole folder tree.  when you logged in with putty, you can use vi, find a how to online and you're good to go, or if you want, you can just download the file using ftp and edit in windows, then reupload it. hope this helpful.

redpig888, 2018-03-20:
Hi, How to make a Serial Port cable? What information can you refer to? Thank you

Permensugus, 2018-03-20:
I'm using arduino as my TTL adapter. you can try googling around. I think it will be complicated to build one, better just buy one if you don't have any, just buy one, I think it's under 5$.

redpig888, 2018-03-20:
Hi  Permensugus,
  Thank a lot!