samtap/fang-hacks #288: Scripts randomnly stopped working on 2 cameras at same time.

shamowfski, 2017-12-22:
https://imgur.com/a/gpoVV

I can manually start all except the rtsp script.  It tells me dev/video0 doesn't exist.  I think my next step would be to reset the cameras and try again, but are these in any way stable ever?  Mine haven't worked reliably in the week I've had them.

samtap, 2017-12-22:
Do they get hot? Some of my cams have run for months and generally RTSP only goes down when I need to unplug it to use a different power socket..

shamowfski, 2017-12-22:
something called iCamera running and hogging /dev/video0?

samtap, 2017-12-22:
That script already exists and should run on boot (unless you disabled it). Also how can iCamera claim it, you said it 'doesn't exist' which is something completely different. Please be clear when reporting an issue.

closed 2017-12-22 by shamowfski