samtap/fang-hacks #289: Unable to downgrade from 3.3.3.25

yhamdoud, 2017-12-27:
I have attempted to downgrade to firmware 3.0.3.56 from 3.3.3.25 multiple times. I have rigorously followed the steps provided by the tutorial in the wiki and have a tried a combination of different power plugs, SD cards and partition layouts. All have resulted in failure. I have skimmed through older open issues and used the tips included to no avail. Before I pour in even more hours I'd like to ask if anyone had success downgrading from 3.3.3.25 using an SD card?

Tempo95, 2017-12-28:
yeah, i have this problem too. Putty not may found port on 115200 and i try any SD card = 64 GB, 16 GB and 8 GB but camera don't see them. Don't see them in mihome and don't see when i try downgrade camera. Please help!

Tempo95, 2017-12-28:
P.S. I try two methdos - SD card and PUTTY, in both cases there were problems

Tempo95, 2017-12-29:
I solve this problem! Just need format in FAT not FAT32 (did not notice that) and download older version MI home.

yhamdoud, 2017-12-29:
Flashing via serial allowed me to downgrade.

closed 2017-12-29 by yhamdoud

scannerfm77, 2018-02-11:
I able to downgrade using 2GB card. It doesn't work with my 16GB card.