samtap/fang-hacks #291: Disabling http server, possible ?

nmaster2042, 2017-12-28:
Hi,

I'm looking for a way to disable the http server to avoid someone else modify the cam parameters.

I looked at the /etc/init.d folder but I can't find the init script of the http server (boa if I undersand).

What I want is to disable it, and be able to enable it again if needed.

Thank you for help.

dasmoover, 2017-12-30:
netstat -tunap | grep httpportgoeshere

this will show you the name of the process and you can use killall PIDhere to kill it.

samtap, 2017-12-30:
The webserver process is called boa. If you use iCamera (the cloud app), it is started automatically and you have to kill it like @dasmoover suggested (not sure if some iCamera or Mi Home feature depends on it).

If cloud is disabled, boa is started by the fang-hacks [script](https://github.com/samtap/fang-hacks/blob/master/bootstrap/fang_hacks.sh#L108) so you could just comment or remove that line to disable it.

closed 2017-12-30 by samtap

nmaster2042, 2017-12-30:
Hi, thank you, I'll try this.