samtap/fang-hacks #293: Rewriting SD-card makes cam unbootable?

bogri, 2018-01-01:
Hi,

I have set up the cam, and everything worked perfectly. I have a 32G card but did not create two partitions. I just used disk makager to burn the image and it just made one small partition. I configured the cam - "Hacks Enabled", "Disable cloud"...and everythin was fine. Then I went for the "Extend data partition" button because I thought it would create the second partition. I got the red text "Reboot required to apply new partition layout!". I did that but still the red text appeared, so I rebooted again. Then (too late) I saw that the resizing is supposed to take quite a while so I guess I messed up the resize2fs process. The cam booted fine and worked fine, but the red message was still there.

Then I thought the best thing to do is to do the whole process all over again . So I took out the card and reformatted it and then re-burned the img (v0.2.0). Now I can't "start" the cam :(. Only orange led is visible. Can't surf to ip/cgi-bin/status. Can't ping the IP. Any suggestions? Can I do some manual editing in the config files?

Br

bogri, 2018-01-04:
Maybe trying resetting the cam according to instructions at "[HowTo: Flash original Xiaomi firmware from sdcard (factory reset)](https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset))" might bring some life into the cam?

bogri, 2018-01-05:
Just tried resetting the cam according to instructions via SD card. Does not work. After some googling I saw in another forum that this method does **not** work for cams with mac addresses starting with 34 - which I have :(. So, I guess it's bricked?

oakbrad, 2018-02-09:
I think did the same thing, however I can access it if I first plug the camera in without the SD, then and put in the card with a fresh burn of the fang-hacks.

As soon as I insert the SD it mounts and I can access the web UI - and then a few seconds later it will 'ding' like it did when I did the initial install. I can then manually disable cloud and restart the scripts.

If I reboot with the SD card in, the LED will go amber - blink amber/blue - blink blue - nothing.

If I take the SD out, boot the camera (wait for blinking blue LED), then plug the card in, it repeats as above with having to re-enable the scripts.

Something's going wrong with mounting the SD card at boot maybe? Happy to provide logs, not sure what is going on.

AntPino, 2020-02-04:
> Только что попытался сбросить настройки камеры в соответствии с инструкциями через SD-карту. Не работает. После некоторого поиска в Google я увидел на другом форуме, что этот метод **не** работает для камер с mac-адресами, начинающимися с 34 - что у меня есть :(. Может быть, следует упомянуть в инструкциях? Итак, я думаю, что это кирпич?

You were able to solve the problem.
I have the same situation.