samtap/fang-hacks #300: No video after a while

cc13com, 2018-01-09:
Hi,

I reactivated my Xiaomi camera yesterday after I while off. I use version 0.2.0 and camera version 3.0.4.20. After I plugged in the camera with power the cgi-bin/status page was running and with VLC I got an unicast-video.

But after a while (a couple of minutes) the video stream was broken but the audio was working. When I reboot the cam all is fine until the couple of minutes.

Any idea?

rayzor666, 2018-01-10:
I have the same issue on multiple cameras, some last longer than others and sometimes the feed comes back on it's own but normally a reboot is needed, I would love a solution for this also

tamamma, 2018-01-15:
Me too same issue, pls solve

waspie, 2018-01-18:
same here and i usually have to do a cold boot at that.

samtap, 2018-01-18:
Does the camera get very hot before the stream goes down? And cools down before it starts working?

I have no way to reproduce this (I've had cams online for weeks without issues), without better feedback no way to fix it. The minimum I need is the stdout/log of the rtsp server when it occurs.

waspie, 2018-01-19:
it's not likely heat related, at least for one of mine. It's out in a garage where its been no more than 30 degrees F for weeks.

rayzor666, 2018-01-19:
Issue #178 with a modified 20-rtsp-server resolved the drop outs for me with a week and a half of stable connectivity to date, the only downside is no audio

waspie, 2018-01-19:
are you referencing waitfor1t's post near the bottom with the revised start script?

cc13com, 2018-01-27:
@samtap How can I help you with support to solve this issue? Where is the stdout/log what you asked before?