samtap/fang-hacks #302: New firmware: 3.3.3.25

dejies, 2018-01-16:
Hi I recently purchased a xiaomi xiaofang smart ip camera. The firmware version is 3.3.3.25.
i tried to apply the hacks , when i put i sd card i can  hear 2 beeps. i can access the page using the url
http://device-ip/cgi-bin/hello.cgi.

but when i tried to access http://device-ip/cgi-bin/status i am getting 404
could you please advice.
Thanks
Dejies

dejies, 2018-01-19:
#samtap ,can you please help
Thanks
Dejies

jcuc, 2018-01-19:
Hello, can anyone know how to downgrade that version of firm? The previous method is not valid.

moskovskiy82, 2018-01-19:
In the same boat... Accidentally pressed the upgrade button

astraforce, 2018-01-19:
I managed to apply the hack  only by serial connection and manually run 0.elf from SSH, even from the firmware 3.2.0.30. It doesn't worked just by inserting the sd card.

dejies, 2018-01-21:
Hi all ,
I was able to downgrade to 3.0.3.56 and applied the hacks using 
https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)#via-sdcard.
The rtsp stream quality was very bad. I tried different configurations but not able to get a good stream.
So I reverted back to original firmware
Thanks
Dejied

jcuc, 2018-01-21:
Dont work for me

moskovskiy82, 2018-01-22:
Easily downgraded the firmware following the link

be3max, 2018-01-23:
I was able to flash 3.0.3.56 (from 3.3.3.25) only when connected by UART

jcuc, 2018-01-23:
how can i connect by UART?

be3max, 2018-01-23:
**jcuc**, you can use this video - https://www.youtube.com/watch?v=OOghPawyIms and https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)

jcuc, 2018-01-24:
Perfect, I was looking for something easier, but it is possible.

djmanosv, 2018-01-27:
can anyone help me to use ”continous recording” option on latest 3.3.3.25 version ? Anyone that has successfuly used this option ? I have a 16Gb micro sd card which should be fine I guess

scannerfm77, 2018-02-08:
I failed downgrade using SD card https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)

moskovskiy82, how did you downgrade? Just put the files at SD card root, press setup button and boot?

4rianton, 2024-06-06:
Finally was able to use this hack. Thanks!