samtap/fang-hacks #303: Cycle recording video like registrator MOD

be3max, 2018-01-23:
Cycle recording video like registrator MOD by be3 

The function of cyclic recording on a hacked camera was missing for me, so i made it.
Registrator mod writes mp4 by 1 minute, files older 4 hours are deleting.

Download FileZilla - https://filezilla-project.org/download.php

connect to camera by SFTP
IP: *YOUR XIAOFANG IP*
login: root
password: ismart12

0) On page htt*://YOUR XIAOFANG IP/cgi-bin/status expand your card to full size - "Expand data partition (/media/mmcblk0p2/)" - Yes. It may take a long time, but once it comes...

1) Script rtsp-registrator.sh copy in folder:
/media/mmcblk0p2/data/usr/bin

Now apply permissions to rtsp-registrator.sh
right mouse button -> Permissions -> 755 -> allow

2) File 30-rtsp-registrator copy to: /media/mmcblk0p2/data/etc/scripts

3) Files registrator_player and registrator to /media/mmcblk0p1/bootstrap/www

4) Reboot camera

5) Go to htt*://YOUR XIAOFANG IP/cgi-bin/scripts -> "Manage scripts"
There will appear new service - 30-rtsp-registrator

6) Recorder video and player are hear - http://YOUR XIAOFANG IP/cgi-bin/registrator

7) You can add registrator button on status page, file:
/media/mmcblk0p1/bootstrap/www/action

Add string:
<button title='Registrator' type='button' onClick="window.location.href='registrator'">Registrator</button>

After:
<button title='View /tmp/hacks.log' type='button' onClick="window.location.href='action?cmd=showlog'">View log</button>

* If no videos are made, set permissions for the directory /media/mmcblk0p2/data/tmp/registrator - permission: 622
**my camera decently hot, to reduce temperature i has decrease quality of rtsp a stream in a file - /media/mmcblk0p2/data/etc/scripts/20-rtsp-server:
snx_rtsp_server -W 960 -H 540 -Q 10 -b 1024 -a >$LOG 2>&1 &
[RTSP-registrator_by_be3_v1.1e.zip](https://github.com/samtap/fang-hacks/files/1655995/RTSP-registrator_by_be3_v1.1e.zip)

cwirek, 2018-02-06:
Is there any possibility to change the rule of deleting recording files for more than 4 hours?

silasb, 2018-02-07:
@cwirek look at `rtsp-registrator.sh` and understand how the `find -mmin +240` works.

bosne28, 2018-03-23:
Could you give me some tips to add audio recording?

be3max, 2018-03-26:
I tried, but couldn't enable audio record :(

RABCbot, 2018-04-08:
I'm testing the script to record, thank you for sharing it!
How can I show the status of the script on the http://192.168.101.120/cgi-bin/scripts page?
I can see the Enable, Run buttons, but there is no status shown
Thanks again

GeitjeG, 2018-04-08:
I cannot find the string
<button title='View /tmp/hacks.log' type='button' onClick="window.location.href='action?cmd=showlog'">View log</button>

In the "action" file. So where should I put the 
<button title='DVR' type='button' onClick="window.location.href='registrator'">DVR</button>

?

GeitjeG, 2018-04-11:
Now, I am not able to start the script. I did set the permission of the "tmp" dir to 622. Also I did connect to the cam with Putty, and chmod +x the rtsp-registrator.sh and 30-rtsp-registrator scripts.

If I run the script i get this:
[image]

RABCbot, 2018-05-23:
GeitjeG can you provide details how you update Jeedom JPG. I have similar problem, I see the rtsp-registrator, but is not running properly.

barqers, 2018-10-02:
> GeitjeG can you provide details how you update Jeedom JPG. I have similar problem, I see the rtsp-registrator, but is not running properly.

I'm also having an issue getting it to run properly - did you manage to get it to run? I hit run and nothing really happens.

peros550, 2018-10-20:
> Aud
> 
> > Did anyone achieve to have audio in the recording?
> 
> I got audio working by making a slight mod to the scripts from here:
> http://bobbyromeo.com/technology/xiaomi-smart-1080p-wifi-ip-camera-rtsp-streaming-hack
> 
> Changing the ffmpeg command line in the actions script with the following:
> `$($FFMPEG_PATH/ffmpeg -nostdin -rtsp_transport udp -i rtsp://localhost/unicast -c:a ac3 -c:v copy -hls_time 60 -hls_list_size 0 $RECORDINGS/$filename 2>/dev/null 1>&2&)`
> 
> This uses the Ac3 codec at 8000hz, and in my testing utilises barely any additional cpu.

Because it was not clear to me which file should be altered, are you suggesting to edit the rtsp-registrator.sh ?

bshaw-au, 2018-10-20:
No - as I said this line is a modification to the actions script- but not the default action script, the action script originally modified by bobbyromeo available from the link I provided.

I have not tested the rtsp-registrator scripts - but it is also relying on ffmpeg so you could take a similar approach:

maybe something like this:
`/media/mmcblk0p2/data/test/ffmpeg/ffmpeg -nostdin -rtsp_transport udp -i rtsp://localhost/unicast -c:a ac3 -c:v copy -hls_time 60 -hls_list_size 0 /media/mmcblk0p2/data/tmp/registrator/${START}.mp4 2>/dev/null 1>&2&)`

I would recommend bobbyromeo's scripts though.

peros550, 2018-10-21:
@bshaw-au many thanks! Your message gave me good guidance. 

For everybody who might be using the rtsp-registrator script,  I just needed to replace this  "-c copy" with "-c:v copy -c:a ac3"