samtap/fang-hacks #305: Some binaries

dvv, 2018-01-24:
### mosquitto_pub/sub 
[mosquitto-client.zip](https://github.com/samtap/fang-hacks/files/1658459/mosquitto-client.zip)

### snx_audio_playback/record 
[snx_audio.zip](https://github.com/samtap/fang-hacks/files/1658461/snx_audio.zip)

Modified to take literal dash as filename meaning standard streams: that allows for network sound servers like:

- remote mic:
```
# tcpsvd 0.0.0.0 12060 snx_audio_record -d s16 -s 48k -f -
```
- remote speaker:
```
tcpsvd 0.0.0.0 11032 snx_audio_playback -d s16 -s 32k -f -
```

An example of network radio:
```
sox -t mp3 'http://retroserver.streamr.ru:8043/retro128' -t raw -c 1 -r 32k -e signed-integer -b 16 - vol 0.2 | nc 192.168.1.238 11032
```

### snx_snapshot 
[motion.zip](https://github.com/samtap/fang-hacks/files/1658478/motion.zip)

Simplified to not maintain `/tmp/snaplist.txt` and just call `./send.sh FILENAME` on every snapshot taken. Eg:
```
mosquitto_pub -h ... -t ... -f "$1"
rm "$1"
exec usleep 250000 # limit to circa 4 FPS
```

RiRomain, 2018-01-24:
Hi @dvv ,

Very nice, I tried the mqtt client and it's working perfectly.

Just wonder, where and how do you call the send.sh code in sny_video_codec.c (if it's there)?
Did you maybe post your modified version online? Or could you? :smile:

dvv, 2018-01-24:
@RiRomain here they are [hacks.zip](https://github.com/samtap/fang-hacks/files/1660627/hacks.zip)
Funny hacking!

RiRomain, 2018-01-24:
Awesome, thanks 👍

toshovski, 2018-01-28:
Can you also provide an example how to connect the local microphone to the remote speaker. The internet radio works perfect, but I would like to transfer the microphone. I played around with arecord, but the sound on the other side is somehow broken:

This is what I tried;
```
arecord -f dat -r 32000 | sox -t raw -c 1 -r 32k -e signed-integer -b 16 - -t raw - vol 0.2 | nc 192.168.1.58 11032
```
I'm very unexperienced with arecord and sox.

dvv, 2018-01-29:
Try
```
sox -d -t raw -c 1 -r 32k -e signed-integer -b 16 - vol 0.2 | nc 192.168.1.58 11032
```
Helped?

dvv, 2018-01-29:
Updated the starting message -- poorman MJPEG streamer spawned

vaidyasr, 2018-02-02:
rsync binary:
[https://sourceforge.net/projects/nmtcsi/files/rsync_xiaofang.zip](url)

mouth4war, 2018-02-07:
@RiRomain Have you integrated this in HASS? Better than the old snx_snapshot solution?

RiRomain, 2018-02-07:
I've only used the mqtt client to send the image via MQTT. I still use the snx_snapshot, but an improved one, posted by KoljaWindeler on this thread: https://github.com/samtap/fang-hacks/issues/87#issuecomment-362955772

MJPEG streaming is of course the best solution, just didn't have time to have a look, and seems I currently get about 1 image / sec, it's already enough for my usage :-)

KoljaWindeler, 2018-02-10:
@dvv you wrote "remote mic to local speaker" (cam mic to laptop speaker) with different port and sample rates, didn't you? this works for me:

CAM:
`tcpsvd 0.0.0.0 12048 snx_audio_record -d s16 -s **8k** -f -`
LAPTOP: (change to the ip of the cam)
`nc 192.168.2.56 **12048** | sox -t raw -c 1 -r **8k** -e signed-integer -b 16 - -d`

Love it! Thanks for sharing!
Kolja

dvv, 2018-02-10:
@KoljaWindeler right. Have fun )

kszere, 2018-04-26:
Not work when rtsp server is run why?

dvv, 2018-04-26:
Presumably because rtsp server grabs the audio device. What's exact rtsp command line?

kszere, 2018-04-26:
'snx_rtsp_server -W 1920 -H 1080 -Q 10 -b 4096 -a >$LOG 2>&1 &'

dvv, 2018-04-26:
`-a` means to stream audio. That's what I meant.

kszere, 2018-04-26:
I need also audio in RTSP stream.

dvv, 2018-04-27:
Then listen to RTSP stream. It's one or the other

kszere, 2018-04-27:
Can XiaoFang use FIFO file in RAM? Need edit `snx_rtsp_server` for source audio and video for custom path.

dvv, 2018-05-03:
@markrennie71 grab mosquitto_pub/sub from the first post and save to the camera. Then follow description under "snx_snapshot"

MCOberle, 2018-05-04:
@dvv Same question as markrennie71. 

This requires snx_snapshot, right?
Where do I need to put it on the camera? Do we need to modify snx_autorun.sh after that?

Can this work together with the RTSP server or is it also one or the other?

MCOberle, 2018-05-04:
I have copied now the following files to /media/mmcblk0p2/data/test/ and made all of them executable:

mosquitto_pub
mosquitto_sub
snx_snapshot
snx_isp_md
send.sh
mjpeg.sh

Then I executed tcpsvd 0.0.0.0 38093 /media/mmcblk0p2/data/test/mjpeg.sh

If I access the cam via a browser I receive a blank page with a question mark indicating the missing image. 

On the camera I receive the following error:

```
Unable to connect (Lookup error.).
Unable to connect (Lookup error.).
Unable to connect (Lookup error.).
Unable to connect (Lookup error.).

```
What am I doing wrong?

dvv, 2018-05-04:
I'll try to describe the way the binaries interact:
- `snx_snapshot` locks video device (hence, _can not be run in parallel to rtsp server_), saves a snapshot to a temporary file (under `/tmp`), calls helper script `./send.sh SNAPSHOT_FILENAME` and removes snapshot file after the helper exits.
- `send.sh` I posted above does publish that snapshot to _your_ MQTT server. You should adjust `YOUR_MQTT_SERVER` and `YOUR_MQTT_TOPIC` in `send.sh`:
```sh
#!/bin/sh
mosquitto_pub -h YOUR_MQTT_SERVER -t YOUR_MQTT_TOPIC -f "$1" -r
exec usleep 250000 # limit to circa 4 FPS
```
- Then we expose a `mjpeg.sh` at a network port via `tcpsvd`. `mjpeg.sh`'s job is to wait for published snapshot file and dump it to stdout. You should adjust `mjpeg.sh` so that `mosquitto_sub`'s host and topic correspond to `YOUR_MQTT_SERVER` and `YOUR_MQTT_TOPIC` in `send.sh`.

NB:
- `send.sh` should be put in the same directory where `snx_snapshot` put.
- `mosquitto_pub/sub` should be under `$PATH` or their full paths should be written in `send.sh`/`mjpeg.sh`.

Please fix your setup and report the result.

MCOberle, 2018-05-04:
I have now put mo`mosquitto_pub/sub` in`$PATH`
An adjusted send.sh to

`#!/bin/sh
# send.sh
mosquitto_pub -h 0.0.0.0 -t image/cam1 -f "$1" -r
exec usleep 250000 # limit to circa 4 FPS`

and

`#!/bin/sh
# mjpeg.sh

#
# GPL
# Copyright 2018 Vladimir Dronnikov
#

cat << _EOH
HTTP/1.1 200 OK
Content-Type: multipart/x-mixed-replace; boundary=--mijpgboundary
Connection: close
Cache-Control: no-cache
Pragma: no-cache

_EOH

while true; do
  echo "--mijpgboundary"
  echo "Content-Type: image/jpeg"
  echo
  mosquitto_sub -h 0.0.0.0 -t image/cam/1 -N -R -C 1
  echo
done`

Now I am receiving the following error:

```
Error: Connection refused
Error: Connection refused
Error: Connection refused
Error: Connection refused
```

Looks like the MQTT Server is not running?

dvv, 2018-05-06:
Right. Its IP should be passed instead of 0.0.0.0.

ktecho, 2018-08-19:
@dvv  Thanks a lot for sharing. I tried the internet radio and it's ultra-cool! :)

Just one question: is there any way I can make the camera work in its original form recording videos on the SD when there is movement or should I provide that functionality myself through scripts?

Thanks!

ktecho, 2018-08-19:
> Presumably because rtsp server grabs the audio device
I can view my RTSP with VLC while listening to internet radio, both at the same time.

dvv, 2018-08-22:
@ktecho I _believe_ you should write videos to SD by yourself. Please consider sharing the solution once you get it work.

skop1986, 2018-11-21:
> ### mosquitto_pub/sub
> [mosquitto-client.zip](https://github.com/samtap/fang-hacks/files/1658459/mosquitto-client.zip)
> 
> ### snx_audio_playback/record
> [snx_audio.zip](https://github.com/samtap/fang-hacks/files/1673257/snx_audio.zip)
> 
> Modified to take literal dash as filename meaning standard streams -- that allows for network sound servers like:
> 
> * remote mic (NB: `snx_audio_record` modified to autostop after a year not 10 seconds):
> 
> ```
> # tcpsvd 0.0.0.0 12016 snx_audio_record -d s16 -s 16k -f -
> ```
> * remote speaker:
> 
> ```
> gpio_ms1 -n 7 -m 1 -v 1
> tcpsvd 0.0.0.0 11032 snx_audio_playback -d s16 -s 32k -f -
> ```
> An example of network radio:
> 
> ```
> sox -t mp3 'http://retroserver.streamr.ru:8043/retro128' -t raw -c 1 -r 32k -e signed-integer -b 16 - vol 0.2 | nc 192.168.1.238 11032
> ```
> Remote mic to local speaker:
> 
> ```
> nc 192.168.1.238 12016 | sox -t raw -c 1 -r 16k -e signed-integer -b 16 - -d
> ```
> Local mic to remote speaker:
> 
> ```
> sox -d -t raw -c 1 -r 32k -e signed-integer -b 16 - vol 1.0 | nc 192.168.1.238 11032
> ```
> ### snx_snapshot
> [motion.zip](https://github.com/samtap/fang-hacks/files/1658478/motion.zip)
> 
> `snx_snapshot` simplified to not maintain `/tmp/snaplist.txt` and just call `./send.sh FILENAME` on every snapshot taken plus call `./alarm MOTION` on motion flag changed. Eg:
> 
> ```
> #!/bin/sh
> # send.sh
> mosquitto_pub -h ... -t image/cam1 -f "$1" -r
> exec usleep 250000 # limit to circa 4 FPS
> ```
> ```
> #!/bin/sh
> # alarm.sh
> exec mosquitto_pub -h ... -t state/alarm/cam1 -m "$1" -r
> ```
> A barebone MJPEG steamer:
> 
> ```
> tcpsvd 0.0.0.0 38093 ./mjpeg.sh
> ```
> ```
> #!/bin/sh
> # mjpeg.sh
> 
> #
> # GPL
> # Copyright 2018 Vladimir Dronnikov
> #
> 
> cat << _EOH
> HTTP/1.1 200 OK
> Content-Type: multipart/x-mixed-replace; boundary=--mijpgboundary
> Connection: close
> Cache-Control: no-cache
> Pragma: no-cache
> 
> _EOH
> 
> while true; do
>   echo "--mijpgboundary"
>   echo "Content-Type: image/jpeg"
>   echo
>   mosquitto_sub -h ... -t image/cam/1 -N -R -C 1
>   echo
> done
> ```
> `snx_isp_md` simplified to just call `./alarm.sh MOTION` when motion detection status changed. Eg:
> 
> ```
> #!/bin/sh
> mosquitto_pub -h ... -t state/alarm/cam1 -m "$1" -r
> ```

Здравствуйте, не могилы вы сделать более подробную инструкцию для обычных пользователей?