samtap/fang-hacks #306: Not able to access ../status

PagesByZ, 2018-01-29:
I can go here: http://my-device-ip/cgi-bin/hello.cgi and I see this:

`snx_autorun.sh
bootstrap
System Volume Information
time_lapse
record
log_34CE00D317D6_s.txt`

But when I go to http://my-device-ip/cgi-bin/status I see this:

> 404 Not Found
The requested URL /cgi-bin/status was not found on this server.

Any idea how to fix it?

be3max, 2018-01-29:
What is your`s xiaofang firmware version?

PagesByZ, 2018-01-29:
It updated as soon as I tried to connect. 3.3.3.6.

PagesByZ, 2018-01-29:
I am using Win32 Disk Imager to write to the SD card. I opened the Image file and tried to verify and i get a Verification failed at sector: 2048. So I am guessing the image wasn't properly written to the SD card?

astraforce, 2018-01-29:
It means you have the firmware 3.3.3.25 which is latest. The hack won't work on that. You have to downgrade the firmware to 3.0.3.56 trough serial connection and after that apply the hack. The SD card may be good.

tomsteenbakkers, 2018-01-29:
I tried to downgrade the camera using this guide. https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)#via-sdcard

But, how do you execute the 0.elf file? I have a telnet connection to the camera, from where I copied the 0.elf to ./tmp, but how do I execute this executable?

samtap, 2018-01-30:
cd /tmp
chmod +x 0.elf
./0.elf

tomsteenbakkers, 2018-01-30:
Thanks for the syntax. I can execute the 0.elf but it's still not working. Any ideas??

```
# ./0.elf
image table size:0x0000003c
index:0
reservations:0x12345678
offset:0x00005e84
size:0x00001000
flash start address:0x00001000
flash end address:0x00001fff

erase section (0x00001000, 0x00002000) don't align
fwupdate end!
#
```

Jesse75, 2018-03-18:
Sorry guys. I've been reading and reading and trying to figure this all out. I'm just confused. Does fanghack0.2.0 work on the US27 firmware 1.8.7.0D_201708091510? Trying to get RSTP server running.

I also have 2 Dome. 13US? Firmware 1.9.1.0H_201708091518. 

What options do I have for either of these?