samtap/fang-hacks #308: Extending cam via gpio and/or i2c?

KoljaWindeler, 2018-01-30:
Hi, I saw that the cam loaded an i2c driver and was wondering if anyone played around with it. 

E.g. adding a port expander. I'm still interested in adding a hardware PIR and maybe some neopixel (could be driven via an i2c connected attiny) or even stronger external ir LEDs).

Will do some testing as soon as my second cam arrives. 

Kolja

samtap, 2018-01-30:
I'm interested as well, but I think most pins are already used to control the sensor, audio, IR filter/leds. Most/all(?) pins are controlled by proprietary drivers of "dubious" quality, i.e. to use the sensor you must enable it with some gpio pin, but you can't disable it. You can use the 'regular' Linux kernel gpio subsystem even while the native modules are loaded, but I haven't played with i2c.
You could also just solder an esp8266 to the uart pins and get a whole range of new options...

KoljaWindeler, 2018-01-31:
So second cam arrived and taken apart. I was a bit surprised that no chip on the pcb is actually connected via I2C. Why would the driver be part of the kernel if the interface isn't used. I tried to dig up the datasheet of the processor but was unable to find one that is longer than 2 pages. Can someone give me a link?
Kolja

everteam, 2018-04-08:
Hello,
I'm very interested too !
Philippe