samtap/fang-hacks #309: Serial connection - cannot login

canthelptinkering, 2018-02-01:
Can someone explain exactly how to login with a serial connection. I have managed to get connect a usb serial adapter and get a putty session going, and can see plenty of activity happening. I am able to type and I seem to get some response after pressing enter, what I cannot do is login with the username and password "root/ismart12" as per the instrucions on this page:

https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)#via-sdcard

I have seen some other instructions on youtube where they tell you to stop the autoboot, but from what I can see you have to hit any key, but it does not seem to help. Any help would be greatly appreciated. Thanks

ril3y, 2018-02-01:
> original-Xiaomi-firmware-from-sdcard-(factory-reset)#via-sdcard

Canthe,

If you are typing and seeing data streaming I suspect you are just missing
the login prompt etc with all of the serial data streaming by.  It is a
pain truly.

There are few things you can do.

   1. Reboot and wait until its a bit more quiet.  Then hit enter a few
   times see if you get the login prompt.  If you do try it as normal.
   2. You can kill the iCamera  app by finding its pid (ps -ef) then kill
   pid you found.  You have about 20 seconds before the router will reboot due
   to the watchdog but you can remove the watchdog by rmmod snx_watchdog (I am
   going from memory)

Hope that helps.

woohzle, 2018-02-21:
Hi,
I'm really new to all this.
I have the same issue. I can see an infinte stream of 
"current *photoStatus status is 1,return value is 1"
the only change to this endless stream seems to be if i hit the keyboard (including control C) or right mouse button.

if I hit "enter" it does give me
"iSmartAlarm login:"
I type "root" and it displays in the terminal window.
hit enter and it asks for "Password:"
type "ismart12" (it doesn't display it in the window) hit enter and it goes to the next line, pauses, then says
"Login incorrect"
all while it is transmitting the endless loop of 
"current *photoStatus status is 1,return value is 1"

according to the terminal window (trawling through what the camera is transmitting), i'm on firmware 3.3.3.25 

Wondering whether they changed the password to the camera from ismart12 with this version of the firmware?

any help appreciated. I have another camera (which I didn't upgrade the firmware) and hacked it and it RTSP streams well). i'd like to get this one to the same state!