samtap/fang-hacks #312: Support for cifs mount

vaidyasr, 2018-02-06:
I was trying to mount my samba/cifs share and I am getting 'No such device' error. I found there is no cifs kernel module enabled, is there a way to build it?

vaidyasr, 2018-02-13:
Finally I have created a cifs.ko module to mount samba/cifs shares. Now I can either write the video directly to the NAS or move the contents of video periodically.

mouth4war, 2018-02-13:
That's great. Could you share?

vaidyasr, 2018-02-13:
Link: https://sourceforge.net/projects/nmtcsi/files/cifs_mount.zip

vaidyasr, 2018-02-13:
Insert slow-work.ko first and then cifs.ko
# insmod slow-work.ko
# insmod cifs.ko

samtap, 2018-02-13:
@vaidyasr Nice work, could you give some instructions how you built the module so people can repeat it and perhaps build other modules as well?

vaidyasr, 2018-02-13:
Under the kernel config, I modified'CONFIG_CIFS=m' and did setup variables ARCH and CROSS_COMPILER manually.
Ran make M=fs/cifs under src directory of kernel and I got the cifs.ko. After doing insmod, I found un-resolved symbols, so I did built slow-work too which fixed the issue.

If anyone have any doubts, please let me know.

closed 2018-10-20 by vaidyasr