samtap/fang-hacks #314: Busybox Full Version v1.28.0

vaidyasr, 2018-02-13:
I have cross compiled busybox with all possible commands, interested can download from:
https://sourceforge.net/projects/nmtcsi/files/busybox_xiaofang.zip

gotttlieb, 2018-02-13:
Please refer to the source code and the way you do it. 
Trusting a ZIP without knowing the source code and the way it has been done  is foolish.

vaidyasr, 2018-02-13:
I have selected almost all options in the 'make menuconfig' and posting the .config may not be a good idea as it is very big. I have been doing cross compiling for the past 9+ years for mips and arm platform.

vaidyasr, 2018-02-13:
Rsync: https://sourceforge.net/projects/nmtcsi/files/rsync_xiaofang.zip
Cifs mount: https://sourceforge.net/projects/nmtcsi/files/cifs_mount.zip/download
LFTP: https://sourceforge.net/projects/nmtcsi/files/lftp_xiaofang.zip

mouth4war, 2018-02-14:
@vaidyasr Do we need to update busybox to use cifs/rsync?

vaidyasr, 2018-02-14:
@mouth4war Not required, the busybox contains missing applets/commands which was not there in the original.

mouth4war, 2018-02-18:
Can't get cifs to work. Syslog shows invalid username errors. Oh well.

vaidyasr, 2018-02-19:
The correct syntax for cifs mount is
mount //192.168.1.1/disk1 /media/mnt -t cifs -o username=xyz,password=xyz

mouth4war, 2018-02-21:
@vaidyasr mount: mounting \\192.168.31.1\下载 on /media/mnt failed: No such device

mouth4war, 2018-02-21:
OK got it to work! Need to insmod after every reboot and I didn't know that. Sheesh.

Sidenote: doesn't seem to work with those chinese characters.

closed 2018-10-20 by vaidyasr

ivanfor, 2020-03-11:
Hi @vaidyasr , could you upload your busybox version agail? It's not available anymore.
Thanks in advance