samtap/fang-hacks #317: My Xiaofang seems to be bricked...

florianproxi, 2018-03-05:
Hello, I have a Xiaofang but I have a problem. When I plug in it is orange fixed then after 15 seconds flashing orange. Then restart and so on.
When I try to connect it to mid-home I am the procedure, at a time I hear a Chinese voice that does not have time to finish, it cuts and restarts. Do you have a solution ?

bigrussianbos, 2018-03-18:
What power supply are you using? If it have ehough power then press the setup button while led is flashing.

AntPino, 2020-02-04:
I have the same problem, the power supply is a standard adapter. I click on the setup button, while the yellow indicator blinks, she begins to speak Chinese and her voice breaks. The camera goes to bootloop again. Help me...

closed 2020-02-06 by florianproxi