samtap/fang-hacks #323: DO it work with Xiaomi xiaofang iSC5 ?

greee-fr, 2018-03-13:
Xiaomi xiaofang Smart 1080P WiFi IP Camera (iSC5 model) can be view here
https://www.gearbest.com/ip-cameras/pp_620813.html
It is very cheap and sounds to have good quality (and 2 can be paired for a larger view!)
Does this hack work with this model ?

CaffeinatedTech, 2018-03-14:
Yeah I just configured one.  Don't update the firmware in the Mi Home app.

bend, 2018-04-14:
Hi, 
I have two Xiaomi xiaofang Smart 1080P WiFi IP Camera, and I tried to apply the hack, but I get a 404 page for the status page. When I go to hello.cgi I can see the snx_autorun.sh script. Any idea what i can do ? The firmware seems to be to the latest one in the mi app.

Thanks

CaffeinatedTech, 2018-04-14:
You must downgrade the firmware.  Check out [this](https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)#via-sdcard) guide.

Marmotte73, 2019-01-18:
I have an iSC5 model. On MiHome i have the firmware 5.6.2.138. I try to downgrade but it seems to be impossible. Have you got an idea ?

watchaaa, 2019-03-04:
> I have an iSC5 model. On MiHome i have the firmware 5.6.2.138. I try to downgrade but it seems to be impossible. Have you got an idea ?

me too.
I have this model and firmware 5.6.2.43 and can't do anything, no downgrade and no cgi-bin status...
The only think that I can do is activate RTSP stream.

I want it for Domoticz and jpg image but not possible for me.... 
Very disappointed

ehno5, 2019-04-10:
> The only think that I can do is activate RTSP stream.

How?

watchaaa, 2019-04-10:
> > The only think that I can do is activate RTSP stream.
> 
> How?

https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/issues/718