samtap/fang-hacks #331: Xiaofang kamera

faradit, 2018-03-28:
Beszívtam, én is, ebből vettem, ami csak Kínában műkszik. Valami segítséget, hogy hogyan tudom itt Magyarországon használni, mielőtt a kukába dobnám.

colombre, 2018-03-31:
állítsd be h kína legyen ar origin ország és akkor működni fog

faradit, 2018-03-31:
Köszi a tippet colombre, de bocsi sajna ez nem oldja meg. Mivel csak a Kínai régión tudod aktíválni, mert eleve máshol meg sem jeleníti a kamerát így ott is van a MI-Home-ben. De sajna akár IOS-on akár Androidon is  olyan ablakot hoz be, hogy csak Kínában működik. És sajna utána olvasva ez a verzió, valóban nem üzemel más kontinensen. De ugye ezt a rendeléskor nem írta az Aliexpress sehol sem, vagy csak nem jó kamerát küldött.  Mert elvileg létezi ebből a verzióból európában működő is. Hát jól beszoptam, talán valaki csak megveszi a beszerzési fél ár alatti 2500 forintért. És hozzáteszem majd ezt egy újhoz, de hogy nem Xiaomi lesz az már biztos. Ha egy világcég ennyire ostoba, hogy a szerverén lekorlátoz egy kamera típust akkor nem érdemli meg, hogy vásároljanak tőle. Egyébként van egy kicsi rendszerem már tőlük, amit bővíteni szerettem volna, egy közel 100 ezres eszközparkkal, amiről szint úgy lettem. Ezek után nekik egy fillért em.

Frazer26, 2018-04-10:
Faradit én tennék vele egy próbát, ha eladó még.

faradit, 2018-04-11:
Mivel sokat nem érek  így vele, így természetesen eladó. Ugyan írtam egy reklamáló mailt, aminek a válaszától sokat nem remélek, de azért azt meg akarnám várni, hátha beígérnek egy cserét. Amire az esély persze úgy is 0.