samtap/fang-hacks #333: night vision

colombre, 2018-03-30:
I realized after new firmware installation that the night vision function became poor. It gives nearly dark picture by using this function. I do not remember well but before it was ok. I think the infra leds are lighting week. Do anybody has the same experience? What is the solution to fix it? Thanks in advance.

masterchop, 2018-06-04:
I have the sample problem if you disable hacks and use a different SDcard no hacks you will go back to normal.

studiwork123, 2019-11-07:
any solution to this problem.. @masterchop  you say undo and redo hack to back to normal  IR working?

masterchop, 2019-11-07:
Yes, go back to normal by disabling the hack in the website, then switch the SDcard with another one.
That will enable the cloud based app from xiaomi, you can check IR is working fine and then re-enable the hacks.

studiwork123, 2019-11-08:
@masterchop  i will try, but.. seems like ir are working but very weak,
from the hack, accesing to ip-camcgi-bin/status and enable/disable ir hack from status,
you can hear click, click sound when activated/desactivated, and you can see how is
working on video image, but seems like its really weak IR.. only objects close to
camera.. less than 1 metre are visible.

masterchop, 2019-11-08:
yes, that happen to me. I think there was a fix also for IR running around the questions here but cant remember i used this along time ago.

studiwork123, 2019-11-08:
@masterchop ok friend. if you find the link, please send me in this tread.