samtap/fang-hacks #334: rtsp server not start

oioilu, 2018-04-01:
Hi,

I have installed script using zip file and see status page on the web browser.
All works except for rtsp server.
On the "Boot Scripts" page "20-rtsp-server" says NOK with red color.

I tried to ssh to the camera and using command to start manually but it tells device in busy.

> /dev #  snx_rtsp_server -W 1920 -H 1080 -Q 10 -b 4096 -a
> Video source = /dev/video1 
> Cannot open '/dev/video0': 16, Device or resource busy

Any other solution I can do to resolve?
Thanks

oioilu, 2018-04-01:
Finally I get it.
On network tab , it should be set to modes other than #Cloud# otherwise Xiaomi script is controlling the video source.

Thanks for your great effort on this magic scripts!

kk246, 2018-04-03:
Try tick the box in the status page that says enable hacks. Save and then restart the camera.

masterchop, 2018-06-04:
You need to stop the cloud apps, thats the only way that work for me.