samtap/fang-hacks #351: Connect to surveillance station synology "failed to load camera function"

gijsje, 2018-05-04:
I did take some time but i have it working now.
It works in Domotocz and i can see the stream in vlc.

I want to setup the camera in surveillance station on the synology nas.
I fill in the ip and port 554 set the camera to user defind H.264 and path to /unicast
When i test the connection it gives me an ok but when i click next i do get "failed to load camera function" (kan de waarde van de camerafuncties niet laden)

Is there something else that i am missing??

Thanks in advance

theirishkiwi, 2018-05-10:
@gijsje I have just setup my camera with Surveillance Station on Synology DS.

My settings were as follows.

IP: as per router
Port: 554
Brand: [User Defined]
Source: /unicast
Username and Password: 
Video Format: H.264
Audio Format: (as per camera settings)

I then clicked "Test Connection" and got a green tick, but the area to the right did not immediately display the video stream and connection failed when I continued to the next screen. But, when I went back and tried again, clicking "Test", getting the green tick and then waiting the stream appeared on the right. 

I then continued through the rest of the setup, leaving mostly the defaults for now, until I understand how it works and then I've tweak the settings.

Cheers
Chris

gijsje, 2018-05-11:
;( not working for me
Did try different things but luck

groggemans, 2018-06-11:
For me it's also working with the settings Chris posted. Are you running the latest version of surveillance station?

gijsje, 2018-06-12:
Still can not connect to survailance station
The Dafang camera works great but not the small one
[image]
[image]
[image]