samtap/fang-hacks #353: Is there a way to run scripts when motion is detected?

masterchop, 2018-05-14:


masterchop, 2018-06-04:
i found that we can apply OpenCV (motion detection, face detection,  to the the RSTP feed that comes from the cam, ofcourse you need another hardware to do it as the camera is super low power, i am using a raspberry pi that works fine the problem is that the camera doesnt have IR vision, for some reason the scripts arent working for me but ill keep testing.

matulekpl, 2019-06-11:
> i found that we can apply OpenCV (motion detection, face detection, to the the RSTP feed that comes from the cam, ofcourse you need another hardware to do it as the camera is super low power, i am [...]

Did You managed to run IR vision??

masterchop, 2019-06-11:
yes, sorry.  I am doing OpenCV and everything already.

closed 2019-06-11 by masterchop

Dylantje, 2019-06-22:
Dear.. Do you have a how to? Or perhaps what home automation system u use? I will try to make a motion sensor working with fanghack