samtap/fang-hacks #358: RTSP server recording always hangs

sedmicha, 2018-06-01:
I'm having some issues with the rtsp server. Whenever I try to record the stream, after a few hours or sometimes even minutes the recording just hangs. I've tried multiple different programs to record the stream (ffmpeg, live555 openRTSP, gstreamer, vlc) and they all have the same problem, and they record other rtsp streams fine, so I'm assuming the issue must be with the rtsp server on the camera. I've tried fiddling around with different settings on the server too and I honestly don't know what else to try. Any idea what could be causing this and how to resolve it?

masterchop, 2018-10-15:
Same problem for me i havent been able to find a way to reconnect it just hangs did you find a solution

masterchop, 2019-05-05:
its the recording app, vlc always works on 1080p streaming, running at 10FPS.

elig0n, 2019-10-02:
Does normal play stream works ?