samtap/fang-hacks #363: how to know if my camera is compatible with fang hacks

lpt2007, 2018-06-06:
My camera have telnet 23 open port.

I can connet to camera telnet with user root without pasword.

uname -a
[root@Ingenic-g1_1:www]# uname -a
Linux Ingenic-g1_1 3.10.14 #36 PREEMPT Mon Jul 31 18:07:54 CST 2017 mips GNU/Linux

lpt2007, 2018-06-06:
system type           : mango
machine                       : Unknown
processor             : 0
cpu model             : Ingenic Xburst V0.0  FPU V0.0
BogoMIPS              : 858.52
wait instruction      : yes
microsecond timers    : no
tlb_entries           : 32
extra interrupt vector        : yes
hardware watchpoint   : yes, count: 1, address/irw mask: [0x0fff]
isa                   : mips32r1
ASEs implemented      :
shadow register sets  : 1
kscratch registers    : 7
core                  : 0
VCED exceptions               : not available
VCEI exceptions               : not available

Hardware              : isvp
Serial                        : 00000000 00000000 00000000 00000000

Bechrissed, 2018-06-24:
In my experience all versions are compatible. On some devices downgrading to a certain firmware is advised (by me)

lpt2007, 2018-06-24:
I put image to sdcard and put it to camera but nothing hapened?!?