samtap/fang-hacks #365: I cant get ANYTHING to work on Xiaomi Xiaofang

Tramsrepus, 2018-06-15:
I have spent 15 hours trying a few instructions how to get it work with Google Drive and different hacks but my camera seems to be immune to everything i through on it. I can't access it via Telnet, Putty nor access it via http.  I am a newbie but I have rooted a lot of phones and updated firmwares etc, but the Xiaomi Xiaofang Camera seems to win this time. I need help if some can point me in the right direction.

Thanks

almostserious, 2018-06-26:
Can you confirm 2 things: 
1. The Model Number ( Do you have the 1S Version or the normal one?)
2. Firmware currently installed

Tramsrepus, 2018-07-05:
Hi Almostserious and thank you for helping me.

1, The cameras are Xiao Fang newest model with the 1080P sensor. I dont have the camera at my place right now so i cant read the sticker on it.
2, The firmware is the latest 3.3.4.122

Thanks, Trams

almostserious, 2018-07-05:
Hi. You have the correct version of the camera for this hack. However, you probably need to downgrade your firmware before you can apply this hack.
https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)#via-sdcard

Tramsrepus, 2018-07-08:
IT WORKED :-D Thank you very much Almostserious.

Do you have any idea how I can access my 4 cameras on a SIM-mobile WIFI router from outside my network? Dynamic DNS does not work on SIM-cards sadly. I also have a Raspberry P3 laying around but i have no clue how to use it and and if it can be useful in this scenario :-D

iCarl, 2018-09-11:
Install Homebridge and homebridge-ffmpeg if you're an iPhone user