samtap/fang-hacks #366: Downgrade firmware 5.6.2.35

Daylights, 2018-06-19:
I have a 2018-03-02 board which does not have the obvious 3.3v,rx,tx,gnd pin holes, but 3 other holes that seem to be rx,tx,gnd. I've managed to get a serial connection and it's telling me it's firmware version 5.6.2.35. This means I cannot add it to mi home outside China (cannot initialise), I cannot downgrade the firmware with the sd-card/setup pin method and when I try to downgrade it using terminal, it's telling me an error:

0.elf: line 1: syntax error: unexpected word (expecting ")")

Any idea how to get it downgraded?

Brklnbnc, 2018-06-20:
Does the MAC address start with 78? I think I have the same problem with a new Xiaofang camera (label on package says manufacture date 2018.03) and can also not downgrade the firmware.

Daylights, 2018-06-20:
No, mine starts with c8. But it looks like these are the new versions with a T20 processor (also used in the WyzeCam V2), you should see something like this using serial monitor:

Board: ISVP (Ingenic XBurst T20 SoC)

Another repo is build for this:

https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/blob/master/hacks/cfw/xiaofang/cfw-1.0.bin

I can see the serial monitor searching for demo.bin. So the idea should be to download this file to a FAT16/32 formatted SD, rename it to demo.bin, place it in the camera, boot it while holding the setup button.

When I'm home, I'll test this.

Daylights, 2018-06-20:
^^ works like a charm!

Brklnbnc, 2018-06-21:
Yes, thank you so much for pointing me in the right direction! I have installed Dafang-Hacks, and it's working great.

sanjeewasam, 2018-06-22:
Thanks all. I also have a unit with MAC address start with 78 and firmware 5.6.2.42. I could config Mi Home app but could not see images in the app though it was on line. I tried various other methods but no luck. 

Instructions here worked for me. Fortunately the little instructions here was enough as I did the same (I think the files are also same) for my Defang camera and there I have done the hard work

Pochi07, 2018-08-05:
Hi, I have a xiaofang camera that starts with 78 and firmware 5.6.2.35. I tried to do what you said, but it doesn't work, probably because my english is so bad and I did something bad. Please could you explain the process in a easy way. Thanks.
What I did is to copy cfw-1.0.bin file to a sd fat32 card and rename it to demo.bin, and also copy the content of the second link (bin, config, controlscripts,...) to the same sd card. I  placed it in the camera and booted it while holding the setup button.
Now it seem is continuesly rebooting in a loop.

fanatic1, 2018-09-19:
Hi, 
I have the same issue with firmware 5.6.2.42. Could anyone be so kind and write little more detailed instruction how to deal with firmware_mod. 
I renamed the cfw-1.0.bin to demo.bin and put all the content from https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/tree/master/firmware_mod to the SD card and then hold the setup button for about 5 sec, until the light start to blink yellow. Then the camera reboot ( probably ) and after that I have only quick blinking yellow light. 
I will appriciate any help. 
Thanks in advance.

markluca, 2018-10-01:
> Hi,
> I have the same issue with firmware 5.6.2.42. Could anyone be so kind and write little more detailed instruction how to deal with firmware_mod.
> I renamed the cfw-1.0.bin to demo.bin and put all [...]

Move all files to the SD card. So move Demo.bin AND the 'firmware_mod' files to the root of the card. Edit wpa_supplicant.conf.dist to reflect your wifi details and then insert the card and follow the reset instructions.

fanatic1, 2018-10-03:
Thanks a lot. But seams there is other issues :(
with your suggestion I receive the following errors in the startup log:

"
Reading configuration file '/system/sdcard/config/wpa_supplicant.conf'
Failed to open config file '/system/sdcard/config/wpa_supplicant.conf', error: No such file or directory
Failed to read or parse configuration '/system/sdcard/config/wpa_supplicant.conf'.
Failed to add interface wlan0
: Cancelling scan request"
after that I renamed "system/sdcard/config/wpa_supplicant.conf.dist to same without .dist
It looks like the web server was started now but asks me for username and password when try to access the http://camera_ip/cgi-bin/status
I found out that the password is root/ismart12. It looks like that is accepted but after that I receive error 404-not found.
Also try to downgrade the firmware with FIRMWARE_660R.bin but with no luck. 
Any help will be much appreciated.

fanatic1, 2018-10-04:
I figure it out. It's working like a charm. Thanks for your help.

zwizardw, 2018-10-12:
How to change time?In config/ntp_srv or ?

tombak82, 2018-10-17:
> I figure it out. It's working like a charm. Thanks for your help.

What exactly did you do then? I also did exactly the same as you did, after logging in, I receive the 404-not found...

zwizardw, 2018-10-31:
Need help. After some time camera pause picture ,but sound keep play in background . How to prevent to still picture ? I want to play in VLC on tablet in my house for unlimited time. Is it posible?