samtap/fang-hacks #368: Quality settings in Shinobi

Bechrissed, 2018-06-27:
Hi,

I have been using two Camera's as home security camera's for a while around the house now with Synology Surveillance station. Although Surveillance station is, in my opinion, brilliant software which works very well for me the Licence prices for additional camera's are plain criminal. They are more than the cost of a brand new NAS and way more than the camera's itself. I wouldn't mind paying for a licence but these prices are just not realistic for home surveillance solution.

Because of this I decided to try and switch to Shinobi (https://gitlab.com/Shinobi-Systems/ShinobiCE/). It's not as fancy as Sureveillance station but it does seem to work. One problem I have is that it seems to take the lowest quality feed of the camera. In Surveillance station I was able to choose from low/medium/high settings, this option isn't available in Shinobi (or I can't find it). Can anybody help me change this so I can get the high quality feed in Shinobi.

I did try to change the quality settings in the Fang Hack RTSP server settings but I don't see any difference when I change the settings. Both in live view and recording.

Can I modify the url loaded or something (maybe: /unicast/medium/ or /unicast/high). Since Shinobi is written in JS I could also try and hack in the requested functionality in myself but I'm not sure how to request the different qualities. Anybody got this working or can help me with more info? Any help is much appreciated.

Thanks in advance!

MrUkleja, 2018-08-13:
Hi,

I am using Shinobi with 4 cameras, Fang being one of them.
I have a simple setup with url: /unicast
I used RTSP server setting so change quality. At the moment i have this: snx_rtsp_server -F 15 -W 1280 -H 720 -Q 10 -b 1024 >$LOG 2>&1 &
I did see difference when changing bitrate. I didn't touch other settings apart from bitrate and fps. 
When I set bitrate too high like over 2048 i started to see artifacts and bed video in short so I needed to lower to 1024. 

I also have Xiaomi Dafang camera ... and I think I will switch Fang to another Dafang ... and use Fang in less important area.

I don't think I helped but wanted to share my case with you.

Bechrissed, 2018-08-13:
Hi, thank you for comment. I also managed to change the quality using snx_rtsp_server. It's working now :)

closed 2018-08-13 by Bechrissed