samtap/fang-hacks #379: Distant RTSP access..?

xreqx, 2018-09-02:
Hello, i can access RTSP locally but not when outside of the network (using port redirection). It works for other YI IP cameras with hacked firmware (older YI Ant mostly) so why not here? Another question, how can i get fang-hack to work withOUT microsd plugged in camera?

a-triant, 2018-09-04:
As about access from outside the network, I think the best and most secure way is to make a VPN connection with the network where your camera is connected. Otherwise if you open it to a public then everyone will be able to connect to it and watch your camera, since it does not have any password protection.

mpseybold, 2018-09-08:
SSH tunnel did work for me: ssh -L 5554:localhost:554 root@cctv1
Then open the rtsp stream from localhost