samtap/fang-hacks #382: Change rtsp port

robyevolution, 2018-09-12:
hi I have 2 xiaofang with the hack installed, I wanted to know if it was possible to change the rtsp port so you could use them both from the external network.

pattyland, 2018-09-16:
You should take a look at your port forwarding config in your router, you don't need to use the internal ports for port forwarding.
I would not recommend to expose the stream directly to Internet anyway!

davidjb, 2019-01-04:
@robyevolution What @pattyland says is absolutely right about port forwarding and avoiding streaming to the Internet.  If you *have* to, you can adjust the `/media/mmcblk0p2/data/etc/scripts/20-rtsp-server` script's call to `snx_rtsp_server` to change the port.  Run `snx_rtsp_server --help` to see all the options but the one you'd use is `-P 1234` to change the port from 554 to 1234.