samtap/fang-hacks #386: Compile v4l2rtspserver

LeVuHoang, 2018-10-03:
Hi all,

We have so many issue with rtsp service (snx_rtsp_server) because it's an old version.

I used Dafang hacks with new Xiaofang camera and it's faultlessly. Dafang hacks use v4l2rtspserver-master service.

Is there anyone able to compile v4l2rtspserver for this old camera or tell me how to do it?

https://github.com/mpromonet/v4l2rtspserver

LeVuHoang, 2018-10-03:
I'm following this instruction and hopefully I can compile a new version
https://github.com/Dafang-Hacks/Main

arifwib, 2018-10-26:
hi, @LeVuHoang any progress on compiling v4l2rtspserver and use it with samtap fang-hacks ?

350d, 2018-10-28:
@LeVuHoang Can't wait for update!

LeVuHoang, 2018-10-29:
I use
snx_snapshot -m -q 100 -n 24 -e 0

and it works perfectly. No crash.

350d, 2018-10-29:
@LeVuHoang wrong thread?

LeVuHoang, 2018-10-29:
> @LeVuHoang wrong thread?
No, nobody tells me a way to compile so I have to use snx_snapshot

350d, 2018-10-29:
@LeVuHoang for video streaming or for image snapshots?

LeVuHoang, 2018-10-29:
I use Milestone Xprotect to capture this as snapshot but you can still see it as video streaming