samtap/fang-hacks #392: Is this working on FW 3.3.4.173?

xpresservers, 2018-11-13:
I wanted to know if this is working for Xiao Fang Camera FW 3.3.4.173?

mrgluek, 2018-11-14:
No, you should downgrade firmware, for me only connecting usb-ttl method worked:
https://www.youtube.com/watch?v=OOghPawyIms

nightgryphon, 2018-11-14:
Please can you share the 3.3.4.173 firmware backup before you downgrade?
Backup manual is here
https://github.com/samtap/fang-hacks/wiki/HowTo-make-backup-snapshot-of-your-device-state-via-the-console

mrgluek, 2018-11-15:
> Please can you share the 3.3.4.173 firmware backup before you downgrade?
> Backup manual is here
> https://github.com/samtap/fang-hacks/wiki/HowTo-make-backup-snapshot-of-your-device-state-via-the-console

Could try on another cam later, why you need it? Just update it from Mi Home app.

nightgryphon, 2018-11-16:
I want to take a look at new scripts and utils. But i don't have Mi Home and don't want to register myself and camera there.

nightgryphon, 2018-11-16:
Also as i read the latest 5.* firmware have troubles with downgrade so i afraid of updating to latest from MiHome

lincomatic, 2018-12-04:
I spent a good part of yesterday downgrading from 3.3.4.173 back to 3.0.3.56. Tried SD method and it didn't work. Finally hooked up serial and command line running of 0.elf kept seg faulting or complaining it couldn't write /dev/mtd1. Finally got it to work with boot to SD with FIRMWARE_660R.bin, but it still complains of corrupted partitions. Anyway, it seems to work OK now. Sorry, I couldn't get a backup of 3.3.4.173, because it was already partially wiped by the time I got the serial hooked up.

BSV2308, 2018-12-05:
Dear user, could you explain step by spet how make a downgrad 3.3.4.173 back to firmware for hack.

BSV2308, 2018-12-06:
> Sorry, I can't tell you the exact procedure, since I had errors and had to retry it over and over again, but the first thing is to connect a 3.3v UART -> USB serial adapter,
> 
> https://www.youtube.com/watch?v=TsuCepL7b5U
> 
> and open up a serial terminal on your PC so you can watch what's going on.
> Then I followed the instructions at:
> 
> https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)
> 
> I first tried the SSH/Serial method, but it had errors, so I tried the SD method, and it also had errors, so I went back to SSH/Serial method, and it finally worked. You can watch the messages in the serial terminal to monitor what's going on

Dear lincomatic
Thank you for your explanations 
Coud you write how copy 0.elf to /tmp

I tried use cp /media/mmcblk0p1/0.elf /media/mmcblk0p1/tmp
 can't stat '/media/mmcblk0p1/0.elf': No such file or directory

lincomatic, 2018-12-07:
sounds like the SD didn't get automatically mounted.. I had that happen sometimes. try:

mount /dev/mmcblk0p1 /media/mmcblk0p1

Notes: 
1) sometimes, for some strange reason, it shows up as /dev/mmcblk1p1
2) you might have to mkdir /media/mmcblk0p1 if the directory doesn't exist
3) do ls /media/mmcblk0p1 to see if the files are there

BSV2308, 2018-12-07:
> sounds like the SD didn't get automatically mounted.. I had that happen sometimes. try:
>
> mount /dev/mmcblk0p1 /media/mmcblk0p1
>
> Notes:
>
> 1. sometimes, for some strange reason, it shows up as [...]

Diar lincomatic
Thank you for your asistens
After :+1:   
mount /dev/mmcblk0p1 /media/mmcblk0p1
cp /media/mmcblk0p1/0.elf /tmp
./tmp/0.elf

I had  Carent version 2.8.1.4

It is a very good result.
Now I will try to do xiao-fang hack

Thank you more

BSV2308, 2018-12-07:
> > sounds like the SD didn't get automatically mounted.. I had that happen sometimes. try:
> > mount /dev/mmcblk0p1 /media/mmcblk0p1
> > Notes:
> > 
> > 1. sometimes, for some strange reason, it shows up as /dev/mmcblk1p1
> > 2. you might have to mkdir /media/mmcblk0p1 if the directory doesn't exist
> > 3. do ls /media/mmcblk0p1 to see if the files are there
> 
> Diar lincomatic
> Thank you for your asistens
> After 
> mount /dev/mmcblk0p1 /media/mmcblk0p1
> cp /media/mmcblk0p1/0.elf /tmp
> ./tmp/0.elf
> 
> I had Carent version 2.8.1.4
> 
> It is a very good result.
> Now I will try to do xiao-fang hack
> 
> Thank you more

I took new problem
After step 4. Place sd-card in device o 
http://device-ip/cgi-bin/status  -> 404 Not Found
http://device-ip/cgi-bin/hello.cgi  -> OpenDir faile

My be ver. 2.8.1.4 not compatible?

mrgluek, 2018-12-07:
Yes, you should upgrade to 3.0.3.56 now with serial connection to camera pins.

BSV2308, 2018-12-07:
> Yes, you should upgrade to 3.0.3.56 now with serial connection to camera pins.
> […](#)

> Yes, you should upgrade to 3.0.3.56 now with serial connection to camera pins.
> […](#)

I'm confused
Because I already gave it
https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)

mrgluek, 2018-12-07:
You got older version 2.8.1.4 with camera reset, now flash 3.0.3.56 and you good.

BSV2308, 2018-12-07:
I did it again and have got "fwupdate end"
I unpluged SD? , reboot
http://device-ip/cgi-bin/status -> 404 Not Found
http://device-ip/cgi-bin/hello.cgi -> OpenDir faile

How check a firmware?

mrgluek, 2018-12-07:
Try in console:
~ # cat /etc/os-release
ISA_VERSION=3.0.3.56

You should keep SD plugged after reboot, scripts run from it.

BSV2308, 2018-12-07:
I can not to understand  
Must I do as in Check the output for "fwupdate end!"
Eject SDCard
Reboot device with: reboot

mrgluek, 2018-12-07:
> 7 дек. 2018 г., в 17:19, BSV2308 ***@***.***> написал(а): I did it again
> Yes, I have same result as you wrote
> Now, I inserted SD and power up CAM.
> I have
> http://device-ip/cgi-bin/hello.cgi -> 1snx_autorun.sh
> 1bootstrap
> System Volume Information
> 2EE298.cramfs.swap
> cramfs-root
> 2EE298.cramfs
> time_lapse
> log_34CE00D3C6E9_s.txt

Try to load fanghacks image to SD card again.

BSV2308, 2018-12-07:
> […](#)

I have the hack
Applying fang hacks scripts...
Running fang_hacks.sh...
Removed stale logfile
Executing script (enabled: 1)
Waiting for cloud apps...
iCamera is running!
Boa webserver is running!
Not linking /media/mmcblk0p1/bootstrap/www/action: already exists
Not linking /media/mmcblk0p1/bootstrap/www/func.cgi: already exists
Not linking /media/mmcblk0p1/bootstrap/www/network: already exists
Not linking /media/mmcblk0p1/bootstrap/www/scripts: already exists
Not linking /media/mmcblk0p1/bootstrap/www/status: already exists
Failed to find hacks in /media/mmcblk0p2/data!
Mounted /media/mmcblk0p2/data
Running startup scripts
Stopping Cloud applications...
  test_UP[630]
  iCamera[628 619]
  miio_client[814]
Module snx_wdt unloaded (0)
Starting Network...
Cloud mode: network unmanaged
Starting ntpd...
Starting telnetd on port 2323...
Starting dropbear on port 22...
Starting FTPd...
Starting RTSP server...
Starting IR script...
Finished

Thank you so much

What me do with new cam fw: 5.6.2.177 and mac: 78:....... for hack?

nightgryphon, 2018-12-10:
regarding downgrade from 3.3.4 i has two issues. First of all the firmware .elf binary file is compiled for older kernel and libraries so it cause seg fault or other incompatibilities when i try to run it. So the only solution seems to work is the upgrading from the bootloader. BUT the bootloader have some incompatibilities with new/large SD cards and it seems the partitions on SD partition table is also sometimes detected not well.
To solve this i found an old 2Gb SD card and formatted it without partitions and put firmware file there. If you have serial terminal connected you can see which filename the bootloader is looking for and also check the if update start or not.