samtap/fang-hacks #396: I cant get ANYTHING to work on Xiaomi Xiaofang

mahbubzone, 2018-12-08:
I have bought Xiaofang 3 days ago. I am tried a few methods but nothing seems to work. 
I think this the new model. It has a setup button on back instead of a setup hole.  
Current Firmware is : 5.6.2.177 
Mac address starts with 78 

I have tried this https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/blob/master/hacks/install_cfw.md 
but not working.  I will be very kind of you if someone help to do it. 

Thanks

boscorelly, 2019-01-03:
It’s a 2018 version, as mine. It doesn’t work like the previous version. 
This doesn’t support older versions of flashing.

mahbubzone, 2019-01-03:
> It’s a 2018 version, as mine. It doesn’t work like the previous version.
> This doesn’t support older versions of flashing.

so nothing will work?

boscorelly, 2019-01-06:
for now, no