samtap/fang-hacks #397: Blurry/fuzzy/grainy image (fungus on lens?)

dudududodododedede, 2018-12-11:
Hi all. Just got 4 of these cameras. 3 work great but 1 has a really blurry washed out image. Comparison of a good camera with the bad one:

bad:
[image]

good:
[image]

I tried the official Xiaomi app, tried different quality settings, tried messing around with rtsp_server settings in fang-hacks but the problem persisted, so I figured it's in hardware. I disassembled the camera, thinking maybe it just needs re-focusing. Getting the glue off the lens threads was a PITA but eventually succeeded. No matter how much I adjust the focus I still cannot improve the image quality. I tried cleaning every element and the sensor itself, compressed air etc, still no improvement. Then I noticed that the lens itself seems to have some sort of etching or deposits. Note that this is on the *inside* part, which faces the sensor (not on the glass that faces the outside). Pictures:

outside:
[image]

inside:
[image]
[image]

No amount of elbow grease is able to clean that thing. Maybe it's corrosion due to humidity or water damage? Or maybe just a defect from the factory? It reminds me of the "Mycelium/fungus spores" problem that SLR lenses have: https://commons.wikimedia.org/wiki/Category:Mycelia_on_photographic_lenses

A replacement lens from ebay costs less than $3 shipped (I went with this one: https://www.ebay.com/itm/M12x0-5-F2-0-Fixed-IR-Board-Camera-Lens-1-56-1-8-2-1-2-5-2-8-3-6-6-8-12-16-25MM/302565807804 )

Has anyone experienced anything like this? What are the odds that the lens I ordered will fit the camera without modification?

dudududodododedede, 2018-12-29:
Just an update on this: the lens arrived, it is a perfect fit, and the image is 100% clear now! Still have no idea what was wrong with the original lens, but happy to have a usable camera now. I put secured the new lens with superglue after adjusting the focus (while the camera was on)

If anyone is looking, this is the version I ordered:
Focal length: 2.8mm
Angle of view: 115°
Size: 17mm(L)*12mm(D)

TurfnSurf, 2019-01-23:
@dudududodododedede:  Thanks for the complete follow-up info. I plan on getting some of these lenses in various focal lengths.  This should allow fitting each camera's field of view to the scene of interest, anywhere from fisheye to telephoto.

mariadjohnson, 2019-05-08:
I found a solution for [Fungus](https://usahealthreports.com/fungus-hacks.html) https://usahealthreports.com/fungus-hacks.html