samtap/fang-hacks #401: Fang isc5 brick, new model?

woody4165, 2019-01-11:
Hi all

I think I've bricked my Fang ISC5.
I bought 2 months ago (!) and now is unusable!!!

I've dissassembled, but I see that it's different from some video on Youtube I saw.
I don't have the pins in the same position and it's not written what they are.

They should be two rounded and one squared, are they?

What should I do now?

[image]
[image]

Thanks

woody4165, 2019-01-12:
Ok, I've found how to connect it to serial.
Squared is RX, then TX then GND.
I'm now connected and it boot

During boot I can see
"Not in sigle board test"

And at the end it stops with
"Not have the sensor driver!!!"

Now I login with root/ismart12 and I get "root login on 'console'"

Then if I insert the sdcard I get 
mmc0: new high speed SDHC card at address ....
mmcblk0: mmc0.... 00000 15.0 GiB
mmcblk0: p1

but if I search for /media I don't see anything and also for /system/sdcard or /system/media

where I can find the sdcard file?

Also if I try to flash from SD using the reset button pressed before pluging in the USB power, It stops at

stevedee78, 2021-03-25:
`I found how to restore original fwI` 
Tell me more pls.