samtap/fang-hacks #402: Is my camera supported? Mac 78

mikkadi, 2019-01-12:
I just got this xiaofang cam, is it supported by fang  hacks?
I tried putting sd card with hacks v2 and after two "clacks" the mi home app works but I can't access status page or hello.cgi. 
I get err_connection_refused error.
Any ideas?
Thanks for help!

[image]

Marmotte73, 2019-01-17:
Idem for me with the iSC5 model. :(

germanpsycho, 2019-01-19:
You've got it up to work with the hack?
I just got the same model..

I think the problem is the installed firmware. Mine came with 5.6.2.77 installed and I am offered 5.6.2.138. Didn't install the update yet..

So, is my current version compatibel with the hack? If not, is it still possible to downgrade the device?

g0053, 2019-01-20:
So, gents, with iSC5 model we can not get this hack work?

kittenarmy, 2019-01-22:
I have one of these newer version with 5.6.2.77 firmware. I noticed before I added it to Mi Home I was able to connect to the camera by Wifi. SSID was `isa-camera-isc5_miap803F` so maybe can pull some of the files of it.

kittenarmy, 2019-01-23:
@mikkadi @Marmotte73 @germanpsycho @g0053  

If your camera has T20 processor like mine then you can use this firmware hack:
https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks

[image]

You must open the camera to tell which processor it has. If it is ARM then it won't work.

jonnykking, 2019-02-01:
I have the T20 processor with the iSC5 model. Should be able to just follow the standard steps then i hope!
Will update soon.

g0053, 2019-02-04:
> 
> 
> @mikkadi @Marmotte73 @germanpsycho @g0053
> 
> If your camera has T20 processor like mine then you can use this firmware hack:
> https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks
> 
> [image]
> 
> You must open the camera to tell which processor it has. If it is ARM then it won't work.

Thanks for suggestion! [went to fetch screwdrivers set =)]

g0053, 2019-02-05:
Thanks @kittenarmy successfully installed...