samtap/fang-hacks #404: Reflash Xiaofang using CH431

Papadoma, 2019-01-20:
I have bricked my xiaofang camera by trying to flash the uboot of EliasKotlyar. I tried the serial access method, but didn't work.
My only choice is to remove the flash memory, hook it up to a CH431 and program it. The flash memory is a GD25Q127.

I have the CH431 and the programmer working. As a binary file I used the one provided in the description of this video https://www.youtube.com/watch?v=DgIP_ey_138
However, the CH431A programmer fails to verify the contents. It says they do not match, yet it looks like it detects and erases the flash successfully. There is no option however for a GD25Q127 chip, so I selected the GD25Q128 instead. Not sure if that is the issue or if the binary file I used cannot be used in such a way.

Any ideas?

Verxuls, 2019-03-06:
Maybe better one esp32 and connect it with flash by SPI.

Papadoma, 2019-03-07:
I don't think there is any issue with the CH431. I think it is the image files as they do not content all necessary parts of the flash like the bootloader.

johnny881386, 2019-12-13:
I have the same issue. I am using CH341A Programmer 1.34 with a CH341A dongle. The target flash is GD25Q127 (printed on the chip) and the program detects it as GD25Q128. The writing process is 100% completed but the verification process says the contect is not match at 50%. I read the new data from the chip and had a binary comparison of the one I read at beginning. 30% of the data are mismatch. The errors are in sector (256 bytes) based location. some sectors are correct and some sectors are wrong. Not sure what happened.

johnny881386, 2019-12-13:
I downloaded a new program tool called CH341A Programmer V1.4 and had a try. The problem goes away.

sergioamr, 2019-12-24:
I am having the same problem, did you try the linux flashrom?

sudo flashrom -p ch341a_spi -n -V -c GD25Q128C --layout rom.layout --image boot -w fullflash.bin
https://github.com/Dafang-Hacks/spiflasher

It flashes the rom but it doesn't boot. I will try the CH341A v1.4

sergioamr, 2019-12-25:
Just found that my VCC was not properly in contact, the Ch341A Programmer gave nice errors, the flashrom was saying everything was fine.

eletronicajambeiro, 2020-12-15:
guys, i havent the same trouble and SOLVED with AsProgrammer 2.1.0.13.

Download it and try! This software detect GD25Q127

eletronicajambeiro, 2020-12-15:
i create a account on this forum to help other peoples with same trouble.