samtap/fang-hacks #412: Impossible to run snx-autorun.sh

diabolyk, 2019-03-07:
Hi,

I have just received two xiaofang but it's impossible for me to run the snx-autorun.sh

When i put the sd in the xiaofang i can hear a "bip-bip" but the http://IP-CAMERA/cgi-bin/status and http://IP-CAMERA/cgi-bin/hello.cgi are not working !!!

I have created my sd card with win32diskimager and i have well connected the camera on MiHome App.
When i boot the device i wait until the blue led is fixed and put the sd but nothing

The firmware in MiHome is : 5.6.2.138

Thank you to help me

asharijuang, 2019-03-09:
same problem

diabolyk, 2019-03-09:
nobody to help us ?

i have bought this model cause the others i have are not working.... :'(

asharijuang, 2019-03-09:
i found this, mybe can help. https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks
i think this (fang-hack) only for old mi fang camera.

diabolyk, 2019-03-09:
i will try it and review here if it's working or not.

diabolyk, 2019-03-09:
so it's working, but right now I will try to configure it...