samtap/fang-hacks #413: Can't start RTSP

zork17, 2019-03-18:
I installed the hack on my Xiaomi Xiaofang camera.
I have installed it on 2 previous camers and it worked ok. They are version 0.2.0 of the hack and 3.0.3.56 of the system.
On this camera it is version 1.0 and 3.0.4.9

When I try to select Run for 02-rtsp-server or Start for 03-rtsp-server (see here: https://take.ms/pdjwd) it does not work.
Also, when I press the Network button I get the message: Unsupported command 'network'
I can access the camera interface from the network but not activate RTSP.
What can I do?

This is the log:
Contents of /tmp/hacks.log:
```
Mon Oct 10 14:00:23 GMT 2016 - /etc/fang_hacks.sh: Executing script (enabled: 1)
Mon Oct 10 14:00:23 GMT 2016 - /etc/fang_hacks.sh: Failed to find hacks in /media/mmcblk0p2/data!
mount: mounting /dev/mmcblk0p2 on /media/mmcblk0p2 failed: No such file or directory
Mon Oct 10 14:00:23 GMT 2016 - /etc/fang_hacks.sh: Failed to find /media/mmcblk0p2/data!
Mon Mar 18 09:27:14 GMT 2019 - /media/mmcblk0p1/snx_autorun.sh: Running (device: mmcblk0p1)
Mon Mar 18 09:27:14 GMT 2019 - /media/mmcblk0p1/snx_autorun.sh: Starting telnetd
rmmod: remove 'snx_wdt': Resource temporarily unavailable
```

zork17, 2019-03-18:
I used an old hack.
Now used the newest hack and after a reboot it works.

closed 2019-03-18 by zork17

aphesiia, 2020-03-17:
Where is the newest hack?