samtap/fang-hacks #415: Is there an update to fanghacks_v0.2.0?

TDPFlorida, 2019-03-30:
When I update the firmware to **v3.0.3.56** the hack appears to work but I cannot set to continuously record option in Mi Home.

When I update the firmware to **v3.3.3.25** the hack appears to work but the SD Card is NOT seen

When I update the firmware to **v3.3.4.107** the hack DOES NOT appears to work but the SD Card is seen (in Mi Home) and can set to continuously record…>Advanced Setting>Storage Management

The next firmware update is **v3.3.4.242** but since the hack does not work in previous versions I didn't upgrade to this.

Is there a newer version of fanghacks?

Thanks

TDPFlorida, 2019-03-30:
MiHome 4.14.101

puxxi2003, 2019-03-30:
Here the firmware v3.3.3.25 works flawless, i used MiHome 5.5.0 to update.
I use it with " Disable cloud applications", the SD is working fine.

With the newer firmwares 3.3.4.x, i also had no luck.

[image]
[image]

TDPFlorida, 2019-03-30:
Thanks for the info. As far as I can tell FIRMWARE v3.3.3.25 works flawless for me also. I have updated through Mi Home (iOS) and the latest is  v4.14.102 so I am guessing you are on Android puxxi2003?
Thanks

puxxi2003, 2019-03-31:
Yes, i run the app on a FireStick4K.

On Android the trick is, you have to wait 10-15 min., when updating  from v3.0.3.56 to v3.3.3.25.

When you update, it takes maybe 5 min. and the app show "OPEN" on the update page.
Some time later, it shows a new update to v3.3.4.107.
Now go to the update page and look closely.

If it shows:
Current Firmware:
New Firmware: 3.3.4.107

You are not done,

It must show:
Current Firmware: 3.3.3.25
New Firmware: 3.3.4.107

It reboots during the update process but wait until you see both Firmware-Versions.

That's the only way i got v3.3.3.25 to work properly.

TDPFlorida, 2019-03-31:
I am able to do a "natural" update via the Mi Home APP (iOS v4.14.102) to get the camera to firmware 3.3.3.25. The Fang Hack seems to work fine...at least everything I've tried. When I do more updates via Mi Home to get the camera to FW 3.3.4.107/242. The Mi Home seems to work fine and the files are stored to the SD Card. The problem I have with this is I need some way to get the files off the SD Card because the camera is HIGH and it takes a tall ladder (an hours work plus I have to borrow the ladder) to get to it so it would be nice to remotely access the SD Card (I currently use WinSCP) but in order to access it with this, I need the hack to work (an update that works with 3.3.4.107/242.) That is why I asked if there is an update?
Thanks

puxxi2003, 2019-04-01:
When you use it as security cam, i bought it only to play with it, you may want to upgrade to 3.3.4.242.
In 3.3.4.242 all china restrictions are gone, you can use the app normally, no complains that you are outside china-mainland.

TDPFlorida, 2019-04-02:
90% of why I use it is for fun tinkering with it but I have found the security aspect of it is cool too. I would love to upgrade the firmware and the only reason I don't is I would like to have access to the SD card remotely. Is there an update to the hack or another hack that I could use to access it remotely?

Thanks

TDPFlorida, 2019-04-02:
I discovered all the restrictions are removed with the new FW but I could bypass that by not going above Mi Home v3.12 (which do not have those restrictions as well.)

TDPFlorida, 2019-04-02:
Actually another thing I could use is a way to set/change the watermark time in FW 3.3.4.x. I use fang hacks to do that but when I upgrade the firmware to 3.3.4.x it sets the watermark time BACK to mainland china.

TDPFlorida, 2019-04-02:
When I change the time in the Mi Home app (even with the latest FW and Mi Home) nothing changes?

Thanks

puxxi2003, 2019-04-02:
To set the right time, i use this in rc.local:

ntpd -q -N -p TIMESERVER
sleep 10
modprobe snx_rtc
hwclock -s
hwclock -t
modprobe -r snx_rtc
ntpd -q -N -p TIMESERVER

You may have to switch this:
hwclock -t
hwclock -s

TDPFlorida, 2019-04-03:
Thank you very much but I'm more confused now lol. Are these in a text file on the root of the SD card named rc.local? They look like some king of script commands?

Thanks

TDPFlorida, 2019-04-03:
I see when I image write "fanghacks_v0.2.0.img" to the SD card, it creates a file rc.local in the bootstrap directory. When I open that file with notepad, it looks like some kind of script. Can I just enter this at the end?:

ntpd -q -N -p TIMESERVER
sleep 10
modprobe snx_rtc
hwclock -s
hwclock -t
modprobe -r snx_rtc
ntpd -q -N -p TIMESERVER

What language is this?
I would like to know what each line is doing so is there a place I can learn the syntax?

Thanks

servant74, 2019-04-03:
https://linux.die.net/ is a good place to look up what modules do.

ntpd starts a time daemon
modprobe adds a oadable module
hwclock probably sets the date and time
last mmodprobe removes a module
then it re-starts or syncs the time daemon.

Anyway that is an overview.

TDPFlorida, 2019-04-03:
So if ALL I am trying to do is change the watermark time, I don't really need fang hacks? I also would like to figure out a way to access the SD card REMOTELY (WinSCP is preferred).

Thanks

puxxi2003, 2019-04-03:
NO.
The timestamp only works, when you turn off the timestamp in the app.
i hope, may be it only works when you completely disable the cloud, never tested it.
To create a real timestamp, you have to use "snx_isp_ctl".
Look here:
http://bobbyromeo.com/technology/xiaomi-smart-1080p-wifi-ip-camera-rtsp-streaming-hack/

Time problem: Thats why i use the commands in the rc.local:
+1 Hour Offset Issue
If after adding the timestamp overlay you noticed that the time is +1 hour off, please see the following. It would appear that this is a common problem.
https://github.com/samtap/fang-hacks/issues/78

Here is also a good place for answers, Wyze V1 uses the same hack.
https://github.com/openipcamera/openipc-firmware/issues

Here you will find also answers.
https://github.com/zecoj/fang-hacks-internal
and
https://github.com/GrumpyMeow/XiaomiXiaofangFirmware/issues

puxxi2003, 2019-04-04:
I was wrong, you can manipulate the timestamp.

Try hwclock -s in cmd.

puxxi2003, 2019-04-04:
This script should may (i hope) do the magic. :-)

Disable the 02-ntpd script in fang-hacks.
Copy the 22-ntpd script to your sd-card --> /media/mmcblk0p2/data/etc/scripts/
Enable script and reboot. 
[22-ntpd.zip](https://github.com/samtap/fang-hacks/files/3041752/22-ntpd.zip)

As i said, you may have to change the position of the "hwclock".
I live in germany and we have summer and winter time:
Summertime (now):
  hwclock -t
  hwclock -s

Wintertime:
  hwclock -s
  hwclock -t

puxxi2003, 2019-04-04:
and the right timezone in fang hacks

http://svn.fonosfera.org/fon-ng/trunk/luci/modules/admin-fon/root/etc/timezones.db

puxxi2003, 2019-04-09:
I don't know a download, update only through Mi home.

Before updating, you have to change the network back to cloud in fang hacks and surely disable "Disable Cloud", if you have. Then start the camera, wait a little bit and press "Reset", the yellow led should blink and the Mi app will find your Xiaofang.

prikaovets, 2019-06-27:
Hi Guys, I'm new with hacking this device and I'm kindly asking does is possible hack Xiaomi XiaoFang cam, 3.3.4.242 FW
Motion detection doesn't work and it is most important for me?!

bbrdaric, 2019-07-12:
Hi @TDPFlorida @puxxi2003 how did you get hack working on v3.3.3.25?

I can downgrade to v3.0.3.56 and then upgrade to v3.3.3.25, but on /cgi-bin/status I am getting error 404

madmax2, 2021-05-03:
@puxxi2003 
"Disable the 02-ntpd script in fang-hacks.
Copy the 22-ntpd script to your sd-card --> /media/mmcblk0p2/data/etc/scripts/
Enable script and reboot.
22-ntpd_google.zip"

I did this but the time is +1hr ahead in my video camera OSD.

Do we need to remove the rc.local file that we created with those other commands you said https://github.com/samtap/fang-hacks/issues/415#issuecomment-478842947?