samtap/fang-hacks #416: Current version 5.6.2.77 mac 34:

dennispreutz, 2019-04-09:
Hi
I have an Xiaofang with an mac adress that starts with 34. 
It is currently att firmware 5.6.2.77. 
I have tried to run the hack with the image file "fanghacks v.0.2.0". And it makes a sound like it is ready, but i cant login via http://device-ip/cgi-bin/status. So i suppose it didn't work. What am i doing wrong?

romaduban, 2019-06-23:
you have Xiaomi XiaoFang 1S
so, you can try to use: https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks

Dylantje, 2019-06-24:
the same problem here..
I use the 1s only i cannot get the hack to work..
Did do the wiki etc etc serval times...

Perhaps something wrong in the code?
I did use a old version for the 1s and works fine for long time...
Try the new and not working any more?

romaduban, 2019-06-24:
When I try on my 1s cam it was working well
I did by instruction from 4pda
https://4pda.ru/forum/index.php?showtopic=794144&st=7960#entry77512329

Dylantje, 2019-06-24:
> 
> 
> When I try on my 1s cam it was working well
> I did by instruction from 4pda
> https://4pda.ru/forum/index.php?showtopic=794144&st=7960#entry77512329

mmm Thanks for the link..
Only bricklayer here,,,,,
So rusian language....:-(   Hope you will help me with the google translate error :-)

CloverDroid, 2020-01-02:
@Dylantje ,how is your progress? I just got a xiaofang with MAC 44 and ver. 5.6.2.77 firmware. Did you succeed with fang-hacks?

nedijahij, 2020-01-17:
> When I try on my 1s cam it was working well
> I did by instruction from 4pda
> https://4pda.ru/forum/index.php?showtopic=794144&st=7960#entry77512329

Mine is 5.6.2_0199 so should i tried this too? I am not sure if it is 1s.