samtap/fang-hacks #418: Unable to flash original firmware.../etc/fang_hacks.cfg still there....and doesn't start as original

mikkel75, 2019-04-24:
Hi,
One of my cams got messed up when I by accident (after successful hack) chose wrong wifi settings in web interface.
I have tried to change these but without success - so no I tried to revert to original firmware by running FIRMWARE_660R.bin copied to /tmp/ as described elseware - I have mounted a serial terminal.

```
/tmp # ./FIRMWARE_660R.bin
image table size:0x0000003c
index:0
reservations:0x12345678
offset:0x00005e84
size:0x00001000
flash start address:0x00001000
flash end address:0x00001fff

index:20
reservations:0x12345678
offset:0x00006e84
size:0x002e5c54
flash start address:0x000c0000
flash end address:0x003bffff

index:40
reservations:0x12345678
offset:0x002ecad8
size:0x006d3070
flash start address:0x003c0000
flash end address:0x00abffff

fwupdate end!
/tmp #
```

But after reboot I get this:

```
/tmp # reboot
The system is going down NOW!
Sent SIGTERM to all procesRTL871X: rtw_cmd_thread(wlan0) _rtw_down_sema(&pcmdpriv->cmd_queue_sema) return _FAIL, break
ses
Sent SIGKILL to all processes
RRTL871X: rtw_dev_shutdown
Restarting system.

U-Boot 2011.09 (Nov 24 2016 - 03:23:06)

DRAM:  64 MiB
MMC:   SD Card not detect
mmci_host_init error - -1

SPI FLASH: 16 MB
In:    serial
Out:   serial
Err:   serial
GPIO[2] is high
Hit any key to stop autoboot:  0
roofsr size = 0x6d3070
## Booting kernel from Legacy Image at 00008000 ...
   Image Name:   Linux-2.6.35.12
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    3038112 Bytes = 2.9 MiB
   Load Address: 00008000
   Entry Point:  00008040
   Verifying Checksum ... OK
   XIP Kernel Image ... OK
OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Linux version 2.6.35.12 (fedora@localhost.localdomain) (gcc version 4.5.2 (SONiX GCC-4.5.2 Release 2011-12-06) ) #27 Thu Dec 22 18:48:16 PST 2016
[... 417 lines cut ...]
not in singleboard test
starting pid 619, tty '/dev/ttyS0': '/sbin/getty -L ttyS0 115200 vt100'

Xiaofang3 login: root
```
It does not start as an original cam, and I still see fang_hack files in /etc ??

```
/usr # ls /etc
TZ                   hosts                passwd
Wireless             hotplug              profile
app                  inetd.conf           protocols
config               init.d               rc
dropbear             inittab              resolv.conf
ds_font_table        ir_cut               sdOkFlag
fang_hacks.cfg       mdev.conf            services
fang_hacks.cfg.bak   miio                 shadow
fang_hacks.sh        miio_client          syslog.conf
firmware             mtab                 twowayaudio.conf
fstab                network              wpa_supplicant.conf
group                no_update_list
hostname             os-release

```
Any suggestions??

kirilnedev, 2019-05-12:
I have exactly exactly the same situation as you... in my case the wifi was not starting at all, so I connected via serial, enabled WLAN0, connected it to wifi, got DHCP IP etc... network works... but I am unable to reset it to factory... 
I have another camera which is with stock firmware.. if you know how can I get the firmware from the working one I can clone it

kirilnedev, 2019-05-12:
Ok I managed to get it back to work with Xiaomi MiHome app.
What I did:
1. I edited the fang_hacks.cfg 
   - Changed the DISABLE_CLOUD to 0
   - Changed the NETWORK_MODE to 0
This should enable the cloud software on the camera. 
2. Restarted the camera 
3. Held the SETUP button for 3-4 seconds until I heard the sound... so I can find it in the MiHome app
4. Added the camera to my MiHome App
    At this point I had the camera setup and recognized, but the firmware it was using was not allowing it to be used outside of China. Once I click on it, it shows the normal screen for 1-2 seconds and then popped up the other screen
5. I clicked on the camera and then quickly clicked on the three dots in the upper right part... then went to General Settings and Firmware update .. I had to do this 3-4 times until I got to the latest version (3.3.4.242)
After I installed the latest version it started working normally with the Xiaomi app.
In the meantime I was monitoring through the serial console, and managed to get a link to one of the latest firmware's including the access key so I obtained a copy of the BIN file in case someone needs a newer firmware

untouchable3500, 2020-04-02:
hi, I'm having the same problem you had, do you still have the updated firmware

kirilnedev, 2020-04-02:
Hi there.. not sure if you got the previous e-mail as the attachment was
too big and it got rejected.

Here's the firmware file I managed to obtain from Xiaomi's servers:
https://1drv.ms/u/s!AlQ2nWIRBjnmgp4lnFdIeHY4_o6QfA?e=4F8d6R

I hope this helps you

Kind Regards,
Kiril

esix4rmyy, 2020-10-08:
did you found any solutions? I have the exact problem.

elig0n, 2021-03-01:
> I have another camera which is with stock firmware.. if you know how can I get the firmware from the working one I can clone it

I'm not sure about getting the firmware but you can clone the various partitions of xiaofnag by sending an image of each them (they are at /dev/mtdblockN  see /proc/mtd) over network with `nc` (on sdcard at `data/usr/bin/nc`)and rewrite them to the other device via tftpboot or maybe with sdcard.

elig0n, 2021-03-01:
> Mounting /media/mmcblk0p1
> mount: mounting /dev/mmcblk0p1 on /media/mmcblk0p1 failed: No such file or directory
> Starting boa webserver...
> right_count=3  value=0 last_value=0
> CGI scripts not found in [...]

@mikkel75 It can't read sdcard, did you put it in ?