samtap/fang-hacks #423: Camera restarting periodically

elig0n, 2019-10-02:
After successful boot, there is a minute or two of full capabilities accessibility (rtsp, ssh, http etc) and then the network connection drops and the camera restarts and then all this happens again.

Fanghacks is set to connect via "Cloud" to WiFI.

lines that might be relevant from `kerndebug` :

```
Oct  2 09:24:45 iSmartAlarm kern.warn kernel: RTL871X: rtw_set_ps_mode(wlan0) Enter 802.11 power save - WIFI-TRAFFIC_IDLE
Oct  2 09:24:45 iSmartAlarm kern.warn kernel: RTL871X: rtl8188e_set_FwPwrMode_cmd: Mode=2 SmartPS=2 UAPSD=0
Oct  2 09:24:46 iSmartAlarm kern.warn kernel: RTL871X: rtw_set_ps_mode(wlan0) Leave 802.11 power save - WIFI-LPS_CTRL_LEAVE
Oct  2 09:24:46 iSmartAlarm kern.warn kernel: RTL871X: rtl8188e_set_FwPwrMode_cmd: Mode=0 SmartPS=2 UAPSD=0
```

skinkie, 2020-03-12:
I see a similar effect. I am not connected via cloud. When the camera is successfully connected the connection quality is 88/100. It has been already 4 times I soldered and desoldered the TTL connection. When the camera is out of the cage everything seems to work fine. I suspect it is something like heat.

skinkie, 2020-05-08:
I have an updated. I first have reflashed the camera to the original firmware, but I don't think this was actually the trick. Because I assumed that the fang-ir-control might affect the issue, I disabled this. Yes, the script was already updated to stop the coil. So I effectively enabled the script again, slightly changed it so it will only switch every 60 seconds, and my camera is now running for over 40 minutes.

skinkie, 2020-05-09:
So I redid the flashing of the firmware, and again. Everything runs now for over one hour without issues. I did not apply the hack "again" but I am able to login to the camera and manually start streaming. I am curious what happens inside the _hack_ that makes it break.

elig0n, 2020-05-31:
@skinkie  I have disabled the IR script but it did not help , camera keeps restarting every 30-40 seconds. Similar behavior was reported in #10 but according to @samtap  the latest release should already have it fixed.

skinkie, 2020-05-31:
@elig0n but the latest release is still 2017 right?

elig0n, 2020-05-31:
@skinkie yes

skinkie, 2020-05-31:
@elig0n if I flash the original firmware. The dropbear ssh server is still available, including the fang-hacks scripts. Could you check if you do this, the hanging stops? I am also curious if the wifi drops, or if the entire cpu hangs. I never got it to break when I had connected it to RS232 with an open camera housing.

elig0n, 2020-06-01:
@skinkie I don't flash the device, I only use the supplied image on an sdcard. The device reboots every 30-40 seconds after I start watching the rtsp stream.

skinkie, 2020-06-01:
@elig0n so what I did to get the device "working" again. Where I describe working as: running for more than 30s without rebooting. I followed the procedure to install the latest supported firmware. 

https://github.com/samtap/fang-hacks/wiki/HowTo:-Flash-original-Xiaomi-firmware-from-sdcard-(factory-reset)

I am using the updated ir script, that disables the coil after it switches between variants.

I am starting the rtsp server manually using ssh. I get suffently more "uptime" than 30s of video. But after a while the camera hangs again. I have not yet figured out if this is a wifi issue or a cpu issue.

elig0n, 2020-06-01:
Perhaps your device is damaged by now?

My device boot log show:

    modprobe: remove 'snx_wdt': Resource temporarily unavailable

[so how it can be ](https://github.com/samtap/fang-hacks/wiki/F.A.Q#the-device-reboots-automatically-after-30-seconds)

closed 2021-03-01 by elig0n

elig0n, 2021-03-01:
So after rewriting latest git fanghacks, `e2fsck`'ing and `dosfsck`'ing sdcard partitions I also commented the `if` condition block that would `modprobe snx_wdt` in `/etc/init.d/rcS` according to @gonzoblue [post](https://github.com/samtap/fang-hacks/wiki/F.A.Q#the-device-reboots-automatically-after-30-seconds) and it seems that the restarts are gone.

skinkie, 2021-03-01:
@elig0n very interesting approach. I'll give the new version a test run.

sshaikh, 2023-03-11:
Did this work? I've tried commenting out the if block (including the modprobe) but still get the disconnections.