samtap/fang-hacks #425: hello! not working for ver 3.3.4_0307FW ?

studiwork123, 2019-11-04:
could you confirm that it´s not working for 3.3.4_0307 ver.

and any other easy way to downgrade ?, connecting usb-ttl method it´s dificult to me..

Thanks.

studiwork123, 2019-11-07:
Yes finally works fina.. 
first dowgrade firware via microsd, 
second install hack.

help for others user : try with diferents sdcards fat32,

ReX1983, 2020-05-16:
What is the procedure to downgrade from 3.3.4_0307? Can you also share the link of the target firmware you used?

Thanks

studiwork123, 2020-05-20:
> What is the procedure to downgrade from 3.3.4_0307? Can you also share the link of the target firmware you used?
> 
> Thanks

please read this link.

https://automatedhome.party/2017/12/18/modify-the-xiaomi-xiaofang-camera-to-work-with-home-assistant-and-other-video-recording-software/

ReX1983, 2020-05-20:
Thanks, it worked!