samtap/fang-hacks #439: Xiaofang boot loop

AntPino, 2020-08-12:
Hello, please help!
After installing the hack, I formatted the SD card. And the camera stopped connecting.
When I plug in it is orange fixed then after 15 seconds flashing orange. Then restart and so on.
When I try to connect it to mid-home I am the procedure, at a time I hear a Chinese voice that does not have time to finish, it cuts and restarts. Do you have a solution ?
I also tried to connect through UART, there is a boot cycle, initialization, then a break and all over again.

AntPino, 2020-08-12:
If I try to connect to Mi Home, then I use the standard power supply that came with the camera. And if I connect via UART, then the power is from the computer.

elig0n, 2021-03-01:
formatting is not enough because there are two partitions on the sdcard, so make sure you remove all of it also don't put it in the device. did you try holding the reset button?
please upload UART logs