samtap/fang-hacks #444: Format SD card after flashing image

denbue, 2021-01-25:
Hi,
I'm having troubles to format the SD card I used with fang hacks. All attempts at formatting fail with very weird error messages.

I tried formating on Mac, Windows and Linux with different tools and on terminal. Each produces their own error message and I can't seem to find the pattern. Only guess is that fang hacks applies some unknown filesystem or parameters when flashing which none of the tools can figure out.

Anyone has some ideas?

elig0n, 2021-03-01:
use `fdisk` or some other partition editor to open the device and remove all partitions inside, then create a new partition of your liking (ntfs, vfat, ext...) which fills the whole drive and then format it

TurfnSurf, 2021-03-23:
You can get the official SD card formatter from [sdcard.org](https://www.sdcard.org/downloads/formatter/sd-memory-card-formatter-for-windows-download/). Just accept the license and it will download.

kanonochina, 2021-03-31:
How do you create partitions using the free official SD card formatter?

maxmod, 2021-05-19:
Use this tool 100% working https://bit.ly/3oBO5JO