samtap/fang-hacks #445: Ismartalarm spot bricked?

Prowling123, 2021-02-03:
Hello! I have 3 cameras from ismartalarm. Two of them are the spot+ and they are working very nicely with the fang hack. The other one is a slightly older one ( the spot without the plus and they have a protruding pir sensor). This normal spot ran the fang hack as well, but when i turned on the hacks there were some error messages and now the darn thing doesnt seem to do anything anymore except showing a solid red light. I have tried to press (and hold) the reset button, powercycling it a few times, but all to no avail. I looked into my router page and the device does not show up anymore. 
Did anyone have the same issue? What can I do to fix it?

ptav, 2021-02-07:
Hi Prowling, the issue may be related to the ismartalarm server no longer working, which started 1-2 weeks ago. I'm actually just trying to set it up on a couple of old cameras that I had lying around. However I can't figure out how to set the wifi parameters. Any ideas? Thanks

Prowling123, 2021-02-07:
I don't think so ptav, the other ones still connect when I remove the SD card. This specific one doesn't seem to boot up anymore. I think if you want your cams to work more or less as before, just use the ismartalarm app. The mesh are server should work when you reset them and add them there

SteakSndwich, 2021-02-27:
i am about to try with my ismartalarm cams and want to avoid bricking them. Did you apply the hack to the normal/latest ismartalarm firmware (5.20.0.20 working with the new app provided from meShare), the old one (5.20.0.10) or did you need to flash another firmware (e.g. Xiaomi) first?