samtap/fang-hacks #446: How to uninstall fang hack?

madmax2, 2021-02-09:
There is not mention in the documentation
on how to uninstall fang hack.

Would be good if this was added to the documentation

TurfnSurf, 2021-02-26:
Just remove the SD card.

elig0n, 2021-03-01:
```
Small system modifications are made when you click Apply on the status page:
    A modified sdcard hotplug script is placed on the device to automatically mount ext2 volumes.
    Modified rc.local and rcS scripts are placed in /etc/init.d to enable hacks when the device is rebooted. It also disables copying original files from /root/etc_default to prevent overwriting changes to files in /etc. This poses a risk so always be extra careful when editing files in /etc
    The fang_hacks.sh script and cfg file are placed in /etc.
```
To clean the device and restore it to its original mode one need to undo these operations. 
For example remove all /etc/fang_hacks.* , copy /root/etc_default/rcS and /root_etc_default/rc.local to /etc/ etc...

greensouth, 2021-04-08:
Same problem, can PLEASE someone with a clean system, upload a copy of the unix system so we can at least compare and restore the critical files? 

At least init.d/rc.local and rcS.

Since I did the hack, the QR isn't linking with the app anymore.