samtap/fang-hacks #447: Help Serial connection? Wiring?

stevedee78, 2021-03-25:
What do I have to do to establish a serial connection to the camera?

nariox, 2021-03-30:
Seems like your model might be different, there should be 4 headers. In the one I have, they look like this: https://blog.express-it.nl/wp-content/uploads/2019/01/20190131_130729-2-2.jpg

San3ko, 2022-06-03:
Your model have connections on the back side of ESP, same as mine do.
Here is some good reference: https://github.com/fritz-smh/yi-hack/issues/118#issuecomment-326738029
You will need to solder wires to contacts and use serial-to-usb adapter. 
I have same board, i only used 3 wires (gnd, tx, rx) and power was supplied via usb port.
Hope this helps.