samtap/fang-hacks #449: Flashing blue light

madmax2, 2021-04-30:
I want to flash this firmware, but when I start up the camera it is now just flashing blue light

I try to connect in the mi home app, but get error - connection timeout, retry after checking unobstructed.

how to get the light to turn solid blue?

regan89, 2021-05-03:
@madmax2 I've noticed this has happened to my camera too over the last couple of days, so I wonder if its widespread. It no longer works with the xiaomi app. I'm not currently running fang hacks on it either (but have in the past). I wonder if a factory reset will be enough to fix it...

pangushk, 2021-06-03:
I just used my XiaoFang 1S for the first time after shelving it for 3 years (after opening the box but  not even power it on once over 1000 days!!!)

I noticed the instruction "Boot the device without sd-card, wait until the blue led stops flashing"  would not apply to me, 
as my Fang was flashing blue LED after it successfully connect to my WIFI.  I push the SD Card in anyway and the hack works!

madmax2, 2021-06-03:
my fanghack is working after I flashed it again with the SD card.

The xiaomi app no longer works with this camera though even when SD card is removed..
I think the chinese server has blacklisted it or something..